esphome/feature-requests #1762: I would like to use name_add_mac_suffix in mqtt topic_prefix

stbungaof, 2022-06-10:
`esphome:
   name: mono
   name_add_mac_suffix: true

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password`

It show device name and wifi hostname like this “mono-8d5128” but mqtt only show “mono”
Topic Prefix: ‘mono’
State Topic: ‘mono/switch/mono_relay1/state’
State Topic: ‘mono/switch/mono_relay2/state’

Is there a way for me to change it to
Topic Prefix: ‘mono-8d5128’
State Topic: ‘mono-8d5128/switch/mono_relay1/state’
State Topic: ‘mono-8d5128/switch/mono_relay2/state’

Sorry, I’m not good at English. I hope you can read and understand.

stbungaof, 2022-06-12:
or allow to use templatable on topic_prefix and on_message: topic like mqtt.publish: topic please.

cavamora, 2023-09-19:
really need this too

gunnarbeutner, 2023-11-04:
I ran into this issue while trying to figure out how to use the exact same firmware image for multiple devices without having them overwrite each other's MQTT messages. I ended up with the following workaround for now:

```yaml
esphome:
  name_add_mac_suffix: true
  on_boot:
    priority: 250
    then:
      - lambda: |-
          auto name = App.get_name();
          id(mqtt_client)->set_topic_prefix(name);
          id(mqtt_client)->set_birth_message(mqtt::MQTTMessage{
              .topic = name + "/status",
              .payload = "online",
              .qos = 0,
              .retain = true,
          });
          id(mqtt_client)->set_last_will(mqtt::MQTTMessage{
              .topic = name + "/status",
              .payload = "offline",
              .qos = 0,
              .retain = true,
          });
          id(mqtt_client)->set_shutdown_message(mqtt::MQTTMessage{
              .topic = name + "/status",
              .payload = "offline",
              .qos = 0,
              .retain = true,
          });
          id(mqtt_client)->set_log_message_template(mqtt::MQTTMessage{
              .topic = name + "/debug",
              .payload = "",
              .qos = 0,
              .retain = true,
          });
```

Mrgove10, 2023-11-09:
Thanks @gunnarbeutner for this fix, worked like a charm. This functionality should definity com as standard if you have multiple devices with the same config to manage.

(if you get the error that the ID is undefined, try replaceing ` id(mqtt_client)` by  `mqtt_mqttclientcomponent` )

bdemeny, 2023-11-15:
+1 Please add support for this in ESPHome, being able to have the topic_prefix using unique_id would align with the Home Assistant fix [ Migrate ESPHome unique ids to new format #99451 ](https://github.com/home-assistant/core/pull/99451).
If the topic prefix can be unique, then it will open the  to use similar firmware for multiple devices without any workarounds.

Another big thanks to @gunnarbeutner for the workaround tho!

marceloajunqueira, 2025-01-06:
This is strange, the discovery topic uses the mac suffix, but the state topic "Forgets" it...
I have the same issue because some devices in my house share the same esphome code. 
When I turn on all at same time, they appear all in the HA interface, but only 1 receives the states of all