esphome/feature-requests #1764: Safe Mode-only OTA

nagyrobi, 2022-06-11:
**Describe the problem you have/What new integration you would like**
Add an optional configuration parameter to only allow OTA when in `safe_mode` state.
Benefits, with such an option set to `true`:
- memory freed up. The whole OTA stuff would not load at all during normal operation
- a bit more secure. Device has to be put in `safe_mode` state first, to be able to OTA to it - and putting it in such a state requires first access through the API.

**Please describe your use case for this integration and alternatives you've tried:**
It's known that OTA doesn't work when using with BLE. Thus loading OTA is completely useless, moreover, its contra-productive as it occupies memory in a circumstance when memory is low anyway.

**Additional context**
Config validation should only allow this parameter to be `true` if a `button` or a `switch` from the `safe_mode` platform is present

mmakaay, 2022-06-17:
Any idea on how much working memory OTA takes, when it's not being addressed via the API? Did you investigate on this? Gut feeling says that this is not that much at all. Not dismissing the idea as a whole, but I'm wondering to what extend the memory rationale applies here.

dennisvbussel, 2022-07-05:
I would like to see this feature however I'm not familiar with the BLE component.

In my case I use a custom-component which makes the OTA process instable. It often hangs and times out when a custom-component is handling interrupts. I can imagine that a similar thing is happening when using BLE. 

The workaround for now is to use a button which enters the safe-mode before starting the OTA update on the server. This can be a tedious job when updating a lot of devices hence it can be handy to add this the OTA integration.

Codename-11, 2022-09-08:
I think this feature would be beneficial. I just resolved an issue with a device not being able to OTA update during normal operation as there wasn't enough memory. It took me longer than it should have to deduce that, but going forward it would be nice. Especially should I reproduce and share said device, I'd hate for someone to have to struggle with the same issue.

jum0n, 2023-11-03:
> I would like to see this feature however I'm not familiar with the BLE component.
> 
> In my case I use a custom-component which makes the OTA process instable. It often hangs and times out when a custom-component is handling interrupts. I can imagine that a similar thing is happening when using BLE.
> 
> The workaround for now is to use a button which enters the safe-mode before starting the OTA update on the server. This can be a tedious job when updating a lot of devices hence it can be handy to add this to the OTA integration.

Agree 100% here.  Looking for a way to automate those devices that have heavy BLE usage in my environment and will not OTA with that still running.  OTA works great when in safe mode, so would love a config parameter that would have it enter safe mode before it starts OTA on that device.