esphome/feature-requests #1766: User map for characters in `lcd_gpio` and `lcd_pcf8574`

nagyrobi, 2022-06-12:
**Describe the problem you have/What new integration you would like**

There are different versions of Character-Based LCD Displays, for example the more commons driven by the HD44780UA00 chip English-Japanese or the HD44780UA02 English-European:
[image]
but there are even more, Chinese and Russian versions too.

The problem is that special characters above ASCII appear on various addresses in these screens. For example the `°` symbol is not in the same location. Using `°` in yaml config directly won't render it correctly, `\xdf` is needed for the English-Japanese model, and `\xb0` for the less common English-European.

Now to simply display `°C` you can't just use `return "\xdfC";` because `C` is a hex number, and the C standard does not define the maximum number of them in an `\x` escape sequence, so it is interpreted as part of it. To overcome this, you need to manually hack around with double-double quotes like `return "\xdf""C";` otherwise unexpected data appears on the screen.

The LCD screens (and the `lcd_pcf8574` / `lcd_gpio` components) also support adding a subset of `user_characters` which help with usage of diacritics in many languages.

The other problem is that using all these in strings is rather awkward, for example with the config below  for `Hőmérséklet` we have to use `H\x09m\x05rs\x05klet`, but even worse, for `Főcím` we need to use `F\x09""c\x07m` (again need for a double-double quote).

**Proposed solution**

How about having a `user_map` configuration item where the user could define which characters/symbols have which hex address (based on table above + the `user_characters`) and from then, no more need to care about these in the rest of the config, replacements should be done automatically, no need for escapes and manual taking care of the occurrence of potential hex characters following them):

```yaml
display:
  - platform: lcd_pcf8574
    user_map:
     - °: "\xdf" # in the English-Japanese rom, \xb0 in English-European rom
     - ó: "\x02" # in user_characters
     - ö: "\xdf" # in the English-Japanese rom, \xf6 in English-European rom
     - ő: "\x09" # in user_characters
     - ú: "\x04" # in user_characters
     - ü: "\xf5" # in the English-Japanese rom, \xfc in English-European rom
     - ű: "\x03" # in user_characters
     - é: "\x05" # in user_characters
     - á: "\x06" # in user_characters
     - í: "\x07" # in user_characters
  user_characters:  # example below for Hungarian
    - position: 0 # \x08
      data:  # back arrow
        - 0b00100
        - 0b01000
        - 0b11110
        - 0b01001
        - 0b00101
        - 0b00001
        - 0b11110
        - 0b00000
    - position: 1 # \x09
      data:  # ő
        - 0b00101
        - 0b01010
        - 0b01110
        - 0b10001
        - 0b10001
        - 0b10001
        - 0b01110
        - 0b00000
    - position: 2 # \x02
      data:  # ó
        - 0b00010
        - 0b00100
        - 0b01110
        - 0b10001
        - 0b10001
        - 0b10001
        - 0b01110
        - 0b00000
    - position: 3 # \x03
      data:  # ű
        - 0b00101
        - 0b01010
        - 0b10001
        - 0b10001
        - 0b10001
        - 0b10011
        - 0b01101
        - 0b00000
    - position: 4 # \x04
      data:  # ú
        - 0b00010
        - 0b00100
        - 0b10001
        - 0b10001
        - 0b10001
        - 0b10011
        - 0b01101
        - 0b00000
    - position: 5 # \x05
      data:  # é
        - 0b00010
        - 0b00100
        - 0b01110
        - 0b10001
        - 0b11111
        - 0b10000
        - 0b01110
        - 0b00000
    - position: 6 # \x06
      data:  # á
        - 0b00010
        - 0b00100
        - 0b01110
        - 0b00001
        - 0b01111
        - 0b10001
        - 0b01111
        - 0b00000
    - position: 7 # \x07
      data:  # í
        - 0b00010
        - 0b00100
        - 0b01100
        - 0b00100
        - 0b00100
        - 0b00100
        - 0b01110
        - 0b00000
```
**Please describe your use case for this integration and alternatives you've tried:**
The new [LCD menu](https://github.com/esphome/esphome/pull/3406) component will likely raise the need to find a solution to these problems, as there the number of strings to be used will increase dramatically.

**Additional context**
https://github.com/esphome/esphome/discussions/3547#discussioncomment-2932369

nagyrobi, 2022-06-13:
A very proffessional approach, but wouldn't it be an overkill?

glmnet, 2022-06-13:
If Unicode chars can be directly mapped to rom codes I don't see the benefit of user defining them. Maybe just a config to select the display type.

nagyrobi, 2022-06-13:
The problem is that in the existing roms (EN-JP, EN-EU, EN-CZ and EN-RU) not all the unicode chars are present, many diacritics are missing from all, that's why we have to use the `user_characters` setting and draw our own. And only 8 user chars are possible while there are more missing so the user really needs to compromise and cherry pick what's most useful out of the missing ones.

nagyrobi, 2022-06-13:
> As a workaround you can have a `- include:` in the yaml with a replace function so you don't work too much escaping characters. The render is using lambdas anyway so it would end up being like: `it.print(replace_lcd(...`

Wow that's a great idea. Can you please show me an example?

nagyrobi, 2022-06-13:
I can see https://esphome.io/custom/custom_component.html but that's not tailored towards an on-demand callable function...

glmnet, 2022-06-13:
> > As a workaround you can have a `- include:` in the yaml with a replace function so you don't work too much escaping characters. The render is using lambdas anyway so it would end up being like: `it.print(replace_lcd(...`
> 
> Wow that's a great idea. Can you please show me an example for such a funcion?

put this inside a .h file where the yaml is and use the [`- include:` ](https://esphome.io/components/esphome.html#includes)

```cpp

std::string replace2(std::string str, std::string substr1, std::string substr2)
{
    for (size_t index = str.find(substr1, 0); index != std::string::npos && substr1.length(); 
        index = str.find(substr1, index + substr2.length() ) )
        str.replace(index, substr1.length(), substr2);
    return str;
}

std::string replace_lcd(std::string str) {
    // adapt here
    str = replace2(str, "è", "😀");
    str = replace2(str, "ó", "⭐");
    return str;
}

nagyrobi, 2022-06-13:
@glmnet Thank you very much! Works!
Here's how: created file `char_replace.h` with the following contents (as per example chars above):

```c++
std::string replace2(std::string str, std::string substr1, std::string substr2)
{
    for (size_t index = str.find(substr1, 0); index != std::string::npos && substr1.length(); 
        index = str.find(substr1, index + substr2.length() ) )
        str.replace(index, substr1.length(), substr2);
    return str;
}

std::string replace_lcd(std::string str) {
    // adapt here
    str = replace2(str, "°", "\xdf"""); // in the English-Japanese rom, \xb0 in English-European rom
    str = replace2(str, "ó", "\x02"""); // in user_characters
    str = replace2(str, "ö", "\xef"""); // in the English-Japanese rom, \xf6 in English-European rom
    str = replace2(str, "ő", "\x09"""); // in user_characters
    str = replace2(str, "ú", "\x04"""); // in user_characters
    str = replace2(str, "ü", "\xf5"""); // in the English-Japanese rom, \xfc in English-European rom
    str = replace2(str, "ű", "\x03"""); // in user_characters
    str = replace2(str, "é", "\x05"""); // in user_characters
    str = replace2(str, "á", "\x06"""); // in user_characters
    str = replace2(str, "í", "\x07"""); // in user_characters
    return str;
}
```

Added to the yaml config of the node:
```yaml
esphome:
  ...
  includes:
    - char_replace.h
```

And in the LCD lambda: `it.print(0, 1, replace_lcd("Test óöőúüűéáí °C"));`
In the lcd_menu lambda: `text: !lambda return (replace_lcd("Test óöőúüűéáí °C"));`

nagyrobi, 2022-06-13:
> To me this is a similar abstraction as the `font` is for the pixel-oriented display.

I keep this FR open because your idea is very good. Until then, the solution above is usable in scenarios as described.

glmnet, 2022-06-13:
Ahh too good `text: ` is templatable there

nagyrobi, 2022-06-15:
> `transform` lambda

Would it work to transform multiple character pairs? Or only one?

numo68, 2022-06-15:
> Would it work to transform multiple character pairs? Or only one?

This just saves you the need to remember to call that `replace_lcd` in each and every lambda. You still have to write the translation function, meaning you would leave your include as is and add
```
  transform: |-
    return replace_lcd(x);
```

Then you can directly use your characters like `°` or `ö` both in `lcd_menu`'s yaml and in `snprintf` strings.

nagyrobi, 2022-06-15:
Works, but causes a glitch with the _mark_ characters at the end of the row, they are not displayed anymore at the end of the line.
```
>Hőmérséklet>   
 Páratartalom > 
```

Using this:
```c++
std::string replace2(std::string str, std::string substr1, std::string substr2)
{
    for (size_t index = str.find(substr1, 0); index != std::string::npos && substr1.length(); 
        index = str.find(substr1, index + substr2.length() ) )
        str.replace(index, substr1.length(), substr2);
    return str;
}

std::string replace_lcd(std::string str) {
    // adapt here
    str = replace2(str, "°", "\xdf"""); // in the English-Japanese rom, \xb0 in English-European rom
    str = replace2(str, "ó", "\x02"""); // in user_characters
    str = replace2(str, "ö", "\xef"""); // in the English-Japanese rom, \xf6 in English-European rom
    str = replace2(str, "ő", "\x09"""); // in user_characters
    str = replace2(str, "ú", "\x04"""); // in user_characters
    str = replace2(str, "ü", "\xf5"""); // in the English-Japanese rom, \xfc in English-European rom
    str = replace2(str, "ű", "\x03"""); // in user_characters
    str = replace2(str, "é", "\x05"""); // in user_characters
    str = replace2(str, "á", "\x06"""); // in user_characters
    str = replace2(str, "í", "\x07"""); // in user_characters
    str = replace2(str, "Ó", "O");
    str = replace2(str, "Ő", "O");
    str = replace2(str, "Ö", "O");
    str = replace2(str, "Ú", "U");
    str = replace2(str, "Ű", "U");
    str = replace2(str, "Ü", "U");
    str = replace2(str, "É", "E");
    str = replace2(str, "Í", "I");
    return str;
}
```

numo68, 2022-06-15:
Dang. Right, this of course throws away all the calculations done on the position of the characters in the line so I cannot build the line in one step anymore. That would be relatively easily fixable for anything that is left aligned (labels) or has a fixed position (mark) but if one uses the `°` in the `format` of a `number`, there is no way to know without actually parsing the utf-8.

I am not going to have the time to do anything significant here until the end of the next week so don't rely on anything...

nagyrobi, 2022-06-15:
Further to think is how to handle forward and backward the case when for example a `select` has characters which need to be addressed by hex address, so they look OK in both Lovelace and the LCD... for this I think you need the [font approach](https://github.com/esphome/feature-requests/issues/1766#issuecomment-1153627585) you suggested...

glmnet, 2022-06-15:
> Dang. Right, this of course throws away all the calculations done on the position of the characters in the line so I cannot build the line in one step anymore. That would be relatively easily [...]

yeah, I guess you can test the length of the transformed string

vietagain, 2022-08-31:
i want to make a big font like this [http://forum.cncprovn.com/threads/8948-CKD-Hien-thi-chu-lon-tren-text-LCD-04x20-voi-Arduino-UNO/page2](url) I have code in esphome but don't know how to do next. Can anyone help
 display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x3F
    id: mydisplay
  

    user_characters:
      
      - position: 0
        data:
          - 0b00001 
          - 0b00011
          - 0b00111 
          - 0b01111
          - 0b11111 
          - 0b00000
          - 0b00000
          - 0b00000 
      - position: 1
        data:
          - 0b10000
          - 0b11000
          - 0b11100
          - 0b11110
          - 0b11111
          - 0b00000
          - 0b00000
          - 0b00000
      - position: 2
        data:
          - 0b11111
          - 0b11111
          - 0b11111
          - 0b11111
          - 0b11111
          - 0b00000
          - 0b00000
          - 0b00000
      - position: 3
        data:
          - 0b00100
          - 0b01110
          - 0b01110
          - 0b01110
          - 0b00100
          - 0b00000
          - 0b00000
          - 0b00000
      - position: 4
        data:
          - 0b11111
          - 0b01111
          - 0b00111
          - 0b00011
          - 0b00001
          - 0b00000
          - 0b00000
          - 0b00000 
      - position: 5
        data:
          - 0b11111
          - 0b11110
          - 0b11100
          - 0b11000
          - 0b10000
          - 0b00000
          - 0b00000
          - 0b00000
      - position: 6
        data:
          - 0b10000
          - 0b11000
          - 0b11100
          - 0b11110
          - 0b11111
          - 0b11111
          - 0b11111
          - 0b11111
      - position: 7
        data:
          - 0b00001
          - 0b00011
          - 0b00111
          - 0b01111
          - 0b11111
          - 0b11111
          - 0b11111
          - 0b11111           
          -

nagyrobi, 2022-08-31:
Please edit your post to fix the URL link and mark your configuration example in code quotes.