esphome/feature-requests #1767: Add Whynter ARC-14S Support to Climate IR

aldalen, 2022-06-13:
**Describe the problem you have/What new integration you would like**

I would like support for the ARC-14S standalone AC system in Climate IR.  I have tried other systems, but they don't work. 

**Please describe your use case for this integration and alternatives you've tried:**

Allows for full climate automation of this unit.  It is only controllable via the remote or buttons on the unit.  I have tried SmartIR and other options in the ClimateIR help page. 

**Additional context**

I captured a bunch of IR remote commands from my remote and built a spreadsheet that captures them + derives patterns.  I noticed that all commands came back with "Received NEC" and used the addr/cmd values to build a protocol table. 
https://docs.google.com/spreadsheets/d/14SvjwZCtIWM_aNN50qefxUmNAZliV9ABrFStE_0oLfI/edit#gid=0

I have quite a bit of programming experience and happy to contribute, but a little lost on how to get started with this.

aeonsablaze, 2022-07-13:
I have the ARC-14SH (same unit, but with a heater) and have just finished decoding the signal from my remote. I'm assuming your remote would be pretty much the same. Each command sends the full state of the device in a 32 bit burst, broken down as follows:
```
[000100100] [001] [0100] [10000] [0] [0] [1] [00000000]
static      fan   mode   timer   cf  tmr pwr temp

static: These first nine bits don't change on any command
fan: 001=high
     010=medium
     100=low
mode: 0001=fan
      0010=dry
      0100=heat
      1000=cool
timer: contains the number of hours for the timer with the bits reversed
cf: temperature unit, 0 for celsius, 1 for fahrenheit
tmr: timer toggle, 0 for off, 1 for on
pwr: power toggle, 0 for off, 1 for on
temp: contains the target temperature with the bits reversed
     	if celsius, contains 0-16 (offset from 16 for a range of 16-32)
      	if fahrenheit, contains 61-89 (no offset, actual range)
```

My programming skills could best be described as "enthusiastic amateur", but I've had a peek at the already included protocols and I'm thinking I can take a stab at implementing this one. I'll report back here when I've made some progress. Or utterly failed. Whichever way it goes.

aeonsablaze, 2022-07-18:
I've got support for this unit coded and fully functional. Pull request is submitted and pending.

@aldalen if you want to test in the meantime, you can add this to your config:

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/aeonsablaze/esphome
      ref: whynter-ac-support
    components: [ whynter ]
```

```yaml
    climate:
      - platform: whynter
        name: "AC"
        sensor: room_temp
        use_fahrenheit: true
        supports_heat: false

```

colemickens, 2022-07-22:
Is there any chance the codes are different with the ARC-14S (without the heater)?

I can see messages in the log:

```
[12:54:05][D][climate:010]: 'AC' - Setting
[12:54:05][D][climate:039]:   Target Temperature: 25.00
[12:54:05][D][climate.whynter:163]: Sending whynter code: 0x12280190
[12:54:05][D][climate:385]: 'AC' - Sending state:
[12:54:05][D][climate:388]:   Mode: COOL
[12:54:05][D][climate:393]:   Fan Mode: MEDIUM
[12:54:05][D][climate:414]:   Target Temperature: 25.00°C
[12:54:05][D][climate:385]: 'AC' - Sending state:
[12:54:05][D][climate:388]:   Mode: COOL
[12:54:05][D][climate:393]:   Fan Mode: MEDIUM
[12:54:05][D][climate:414]:   Target Temperature: 25.00°C
```

AND I can use my smartphone camera to see that the IR Transmitter is indeed sending out a singal.

I have the transmitter about 2 feet in front of the AC unit right now, yet the AC unit never seems to act like it's getting anything. Nothing every changes, there's no beeps, etc. Is there anything I should check?

colemickens, 2022-07-22:
My apologies, I have really crappy Dupont cables and a single-LED infrared transmitter. When I fix the wires and aim it very precisely it does work. Thank you!

aeonsablaze, 2022-07-22:
Awesome, thanks for testing! I was pretty sure it would work for the ARC-14S as well but it's good to get actual confirmation.

aldalen, 2022-08-02:
Hey sorry I missed this.  I actually have both a ARC-14S and ARC-14SH.  I'll try this out tonight or tomorrow and let you know!

closed 2022-08-08 by jesserockz