esphome/feature-requests #1770: Binary sensor for whether camera is streaming

jamesmyatt, 2022-06-16:
**Describe the problem you have/What new integration you would like**

I'd like an easy way to expose a binary sensor that represents whether the stream is started or not.

**Please describe your use case for this integration and alternatives you've tried:**

This would make it easy to use HA to (e.g.) switch on a separate light or indicator when streaming starts/stops.

I assume that you can do something with the 2022.6 'on_stream_start' and 'on_stream_stop' stream events (https://github.com/esphome/esphome/pull/3285) and a switch or something 🤷 .

**Additional context**

This is an extension of the original use case for https://github.com/esphome/esphome/pull/3285.

nagyrobi, 2022-06-17:
Try like this:

```yaml
binary_sensor:
  - platform: template
    name: "Camera is streaming"
    id: cam_streaming

esp32_camera:
  name: My Camera
  ...
  on_stream_start:
    - lambda: 'id(cam_streaming).publish_state(true);'
  on_stream_end:
    - lambda: 'id(cam_streaming).publish_state(false);'
```

jamesmyatt, 2022-06-20:
Thanks @nagyrobi, I'll try that

jamesmyatt, 2022-06-20:
That seems to work, except that it should be "on_stream_stop".
Thanks

jamesmyatt, 2022-06-29:
@nagyrobi , if you fix the alternative config, then I'm happy to close this.

closed 2022-06-29 by nagyrobi