esphome/feature-requests #1774: Web Server Component - allow custom order of elements

MagMeister, 2022-06-19:
In web server V1, the order in which elements are displayed in the web interface are determined by the type (sensor, switch, ...) and then in alphabetical order.

Since web server V2 the order is alphabetical.

It would be useful if it were possible to define an order for each element, in the style of a "z-index" and in second, the sorting could then be done alphabetically.

jaggzh, 2023-07-31:
Yeah this is annoying.  Perhaps:
1. Sort type: Defaulting to "alpha by name" for v2 backwards compatibility?
2. Sort type: "index", uses an item's "sort_index: #" value (default 0).  No errors -- components with the same index # receive the same rank and, thus, an arbitrary order.

For optional grouping, higher up hierarchy components could possibly also allow sort_index, where the sort would bring their children together.  If not specified, perhaps a default of -1 would be useful, where they do not receive a sort.  OR we have a "sorted: true/false" on them.

I'm a bit new to esphome config so I'm not sure how compatible this all is with the config philosophy.

bilbs84, 2024-02-29:
For certain, I think this would be a fantastic idea.  If only for testing and development.  I usually interface with all my devices through HA, but when setting up a new device, getting it's calibrations right, etc, I do use the web server component, and it can be quite annoying when you have sensors that should be grouped, be at opposite ends of the list.

I hope something like this does get implemented in a future release.

smarthomedesign, 2024-03-29:
hello all, i guess this idea wan not implemented ?

MagMeister, 2024-12-18:
This has been addressed by the [entity sorting](https://esphome.io/components/web_server#config-webserver-sorting) and grouping in web server V3.

closed 2024-12-18 by MagMeister