esphome/feature-requests #1775: Support IR Remote Climate "Argo Ulisse 13"

coxtor, 2022-06-19:
**Describe the problem you have/What new integration you would like**
Thei IR Climate component could support the argo ulisse 13 climate unit

**Please describe your use case for this integration and alternatives you've tried:**
Integrate the climate unit in esphome

**Additional context**
It is supported already by IRremoteESP8266
https://github.com/crankyoldgit/IRremoteESP8266/blob/77ba26b4d9a53a61b243251aec6d8f7183518613/src/ir_Argo.h
and tasmota 
https://github.com/arendst/Tasmota/blob/381100d8a42b8d142e45e33f09e5a96d6b1581e5/lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/ir_Argo.h

AndBu, 2023-08-26:
I would love to see that as well :)

yoziru, 2024-05-29:
https://github.com/yoziru/esphome-argoclima-ir solves this for the WREM-3 remote