esphome/feature-requests #1776: Delay reset

pencilhead1, 2022-06-20:
**Describe the problem you have/What new integration you would like**

Is it possible to "reset" a delay that is already running.
I have a switch that is running with a delay, I would like the ability to reset the running delay back to 0 seconds if the switch is turned off. It would then be ready with a new 600s delay when it is switched back on.

` # Top (or only) button`
 ` - platform: binary`
  `  name: "${friendly_name} light"`
 `   output: relay1`                     
`    id: pwdr_light                        `
`    on_turn_on:                           `
`     - delay: 600s                        `
`     - light.turn_off: pwdr_light         `
`    on_turn_off:                          `
`    - delay: 0s ` # If switch is turned off at the button, the existing on turn_on "delay" is reset is overwritten (reset) ready for next time switch is turned on 
    #alternative                         
`      - delay_reset ` # function to clear any existing /running delays.
`                                          `
`                                          `
`# Buttons                                 `
`binary_sensor:                            `
`  # Top (or only) button                  `
`  - platform: gpio                        `
`    pin:                                  `
`      number: GPIO16                      `
`      mode: INPUT                         `
`      inverted: True                      `
`    name: "${friendly_name} Button"       `
`    #toggle relay on push                 `
`    on_press:                             `
`      - light.toggle: pwdr_light          `

**Please describe your use case for this integration and alternatives you've tried:**

Currently if the switch is turned off before the delay runs out and then turned back on, the light will turn off after the remainder of initial "turn on" delay event runs out. The "- delay: 600s" doesn't get overwritten by a new "on turn_on" - starting a new 600s countdown.

**Additional context**

pencilhead1, 2022-06-20:
> Have a look at [scripts](https://esphome.io/guides/automations.html?highlight=automations#script-component).
Thanks for showing me scripts. I've got some study to do :)  Do the scripts run on the device or do they need to run on HA?  (still learning). Clearing a delay should (for me) be able to run on the device. I am looking at putting some devices in areas that may not have network access to HA, but still need to run functions autonomously. The ability to reset/clear a delay seems to me like it should be a fairly fundamental function. The same as setting a delay.

Thanks again for the link to scripts.

nagyrobi, 2022-06-20:
> Do the scripts run on the device or do they need to run on HA?

The scripts linked above run on the device itself directly, not in HA.