esphome/feature-requests #1779: Add Waveshare 2.13 inch B72 epaper display support

Tzaphkiel, 2022-06-25:
**Describe the problem you have/What new integration you would like**

I have a **Waveshare 2.13inch e-Paper HAT** display that does not work with *Home Assistant* and *ESPHome*.
My arduino code tests with this display show that the **B72** version of the display should be used.

**Please describe your use case for this integration and alternatives you've tried:**

Using the e-Paper disaply with HA and ESPHome does not allow me to display properly the content.
The text appears with some of the models available but the screen is not completely blanked or clean thereafter.
Looking at the ESPHome waveshare_epaper components code, there are different models available but not the one that I found to be working with my Arduino code: **B72**.

**Additional context**

Code to integrate:
* https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213_B72.h
* https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213_B72.cpp

I am willing (and probably mostly able) to integrate it. I do not feel confortable delivering it to the full project.
I will see if I can start and submit a 'draft' PR to the project at some point.

closed 2024-03-11 by clydebarrow

acfnews, 2024-04-21:
> I am willing (and probably mostly able) to integrate it. I do not feel confortable delivering it to the full project. I will see if I can start and submit a 'draft' PR to the project at some point.

Hi Tzaphkiel,

Can you help me push me in a direction for implementing another display? 
I am a bit overwhelmed, and not sure where to start.
My opened ticket: [new](https://github.com/esphome/feature-requests/issues/2696)

Tzaphkiel, 2024-04-21:
Hi @acfnews, I suggest going the same route I did and prove this display works with one of the drivers available gxepd2 repo above. 

I did that by using an arduino or esp standalone with the arduino ide and getting it to work first; not through esphome. 

You then can report about it in your ticket to help the devs or follow the integration done in this pull req!