esphome/feature-requests #1787: Thermostat: preset option to store changed aspects and restore them when switching

Alex9779, 2022-07-05:
**Describe the problem you have/What new integration you would like**

https://github.com/esphome/esphome/pull/3298 will bring the possibility of default and custom presets to the thermostat component.
This request is meant to be the follow up for https://github.com/esphome/feature-requests/issues/918.
As discussed in this request there is a need in certain use-cases to have presets which remember changes to their aspects and restore those changed values when they are re-activated in contrast to the normal behavior of setting the default values.

Regarding the questions in https://github.com/esphome/feature-requests/issues/918#issuecomment-1175052433 I am not really sure how to handle this and what this means for other aspects than temperature because I am just using a it for heating, no cooling and no fans and other stuff.
Using it with a 'heating_action' the config requires me to set a 'default_target_temperature_low' in the default config for the component and in any preset.
For the moment I think an option for the preset might solve this, a "storing" preset could store all the aspects for which it has a default value and disregard any other so switching to it will keep the current value for that aspect and restore the values of all stored.
But this would not work with the current logic of the config as I am not able to configure a preset without a 'default_target_temperature_low' so this check would have to be weakened.

**Please describe your use case for this integration and alternatives you've tried:**

I need this for a heater control where I want to switch to a preset when anyone is at home and switch back. But I don't want to have the default values restored when switching back.

**Additional context**

Alex9779, 2022-08-02:
I had a little time and created "something" which works as far as I can see for temperatures.
I skipped mode, fan and swing stuff for now.
I didn't create a PR yet, maybe someone can have a look and we can discuss...

https://github.com/Alex9779/esphome/commit/9737179e1c37324affbf3555e6fb516296d1882e

nagyrobi, 2022-08-02:
If noone from the devs reacts here, you should create a PR (maybe as draft), it will draw attention quicker.