esphome/feature-requests #1789: Bedjet V2 support

jcustenborder, 2022-07-08:
**Describe the problem you have/What new integration you would like**

It looks like support for the v3 version of the [BedJet](https://bedjet.com/) has been added. It would be amazing to have support for the older v2 version as well. It seems to use a similar protocol to the v3.  

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jhansche, 2022-07-08:
The protocol is relatively similar, though the status packet is a different format. I think we can support this by switching the codec object either via config (first iteration) or by detecting it at runtime.

jhansche, 2022-07-09:
I've been able to reverse engineer most of the protocol itself, particularly the incoming status packet, and the outgoing command/scene packets. 

However, the transport layer is actually not as similar as I thought it would be. The BedJet V3 uses a series of BLE GATT service characteristics to communicate, but the V2 appears to connect to the device via Bluetooth SPP. Unfortunately I don't really have an easy way to test this, but I can try some things.

@jcustenborder do you think you'd be able to get `nRF Connect` up and running (there are iOS and Android apps, but I've only used the Android app), and capture a log from scanning to pairing/connecting, and receiving realtime status updates? This might help me with emulating the V2 device for testing, before I have something for you to try.

jcustenborder, 2022-07-19:
@jhansche Sorry buddy! I didn't see a notification. I'll work on this and get something for you.