esphome/feature-requests #1790: Add `on_timeout` handler in `wait_until`  action

markormesher, 2022-07-08:
**Describe the problem you have/What new integration you would like**

Since [2021.10.0](https://esphome.io/changelog/2021.10.0.html#wait-until) ESPHome supports a `timeout` property on the `wait_until` action, which addresses half of #1339. I would like to extend `wait_until` further with an `on_timeout` property to allow actions to be run if and only if the timeout expired.

**Please describe your use case for this integration and alternatives you've tried:**

My use case is waiting for responses or state changes when communicating with other devices; if no response is received within some time period it usually indicates an error that should be handled, rather than waiting forever.

The new `timeout` property goes a long way towards enabling this, but it still requires some extra work to determine whether the condition was achieved or the action timed out (essentially just checking the condition again).

**Additional context**

Prior context: #1339

jeroen85, 2023-04-18:
Exactly what I am looking for :-)

illigtr, 2023-09-17:
Absolutely need this. I have not been able to find any alternative way to handle this.