esphome/feature-requests #1791: Update espressif32 from 3.5.0 to 5.0.0 to fix C3 and S3 boards

bruvv, 2022-07-09:
**Describe the problem you have/What new integration you would like**
The current espressif32 is very old (5 months) and therefor there are issues with C3 and S3 esp32 boards.
Currently v5.0.0 is just released: https://registry.platformio.org/platforms/platformio/espressif32/versions

**Please describe your use case for this integration and alternatives you've tried:**
I tried updating myself but I failed

**Additional context**

ma-lalonde, 2022-09-15:
Now at 5.1.1 but still. Supporting to add several Ethernet PHYs compatibility

hapklaar, 2022-10-01:
Can't you choose the version to use in the yaml?

https://esphome.io/components/esp32.html

p2baron, 2023-01-24:
> Can't you choose the version to use in the yaml?
> 
> https://esphome.io/components/esp32.html

You can only choose between recommend, latest and dev.
Dev currently gives you version 5.2.0

bruvv, 2023-01-24:
@p2baron you can it is in the URL:
```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
    version: 2.0.0
```

closed 2023-01-24 by bruvv