esphome/feature-requests #1792: Add Trigger Grow Fingerprint Reader on_scanning

AxokGit, 2022-07-10:
**Describe the problem you have/What new integration you would like**
I would like to add a trigger for fingerprint readers that triggers when a finger is placed on the reader and verification is in progress.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to add an indication to my system that the finger has been detected and is being verified. This way, on_finger_scan_matched or on_finger_scan_unmatched can be triggered afterwards to give more information about the laid finger.

**Additional context**
Having already been able to do this program in C++ Arduino, I know that it is possible to do this. Moreover, ESPHome displays in the logs that a finger is placed before having the result of the verification. So I think the integration is possible.

alexborro, 2023-09-08:
Yeah, that is possible and also easy to implement. I will do it.

alexborro, 2023-09-15:
It is implemented by https://github.com/esphome/esphome/pull/5396

alexborro, 2023-09-27:
I would like to ask to the folks that can test it, to do it with my branch using external components and post the results in the opened PRs listed below.

The snippet for the external components is:

```
external_components:
  - source: github://alexborro/esphome@ext_comp
    components: [ fingerprint_grow ]
```

Just put it before the `fingerprint_grow:` declaration. 

https://github.com/esphome/esphome/pull/5365
https://github.com/esphome/esphome/pull/5396