esphome/feature-requests #1795: Airthings Wave (non plus version)

janchrillesen, 2022-07-15:
**Describe the problem you have/What new integration you would like**
I have an Airthings Wave (non plus model) - similar to the Wave plus, but only checks radon + temp and humidity.
It's detected by airthings_ble and ESPhome sees it as an airthing device and also reads the serial number. However the data received can't be decoded

**Please describe your use case for this integration and alternatives you've tried:**

I tried to add it as a plus model, and this is the result (the config used is identical to the example at https://esphome.io/components/sensor/airthings_ble.html) - removing the config lines for co2 and tvoc doesn't make any difference

[12:42:00][D][ble_client:047]: Found device at MAC address [58:93:D8:8C:4D:D2]
[12:42:00][I][ble_client:085]: Attempting BLE connection to 58:93:d8:8c:4d:d2
[12:42:01][I][airthings_wave_plus:015]: Connected successfully!
[12:42:02][I][ble_client:170]: Service UUID: 0x1800
[12:42:02][I][ble_client:171]:   start_handle: 0x1  end_handle: 0x9
[12:42:02][I][ble_client:383]:  characteristic 0x2A00, handle 0x3, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A01, handle 0x5, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A04, handle 0x7, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2AC9, handle 0x9, properties 0x2
[12:42:02][I][ble_client:170]: Service UUID: 0x1801
[12:42:02][I][ble_client:171]:   start_handle: 0xa  end_handle: 0xa
[12:42:02][I][ble_client:170]: Service UUID: B42E4A8E-ADE7-11E4-89D3-123B93F75CBA
[12:42:02][I][ble_client:171]:   start_handle: 0xb  end_handle: 0x16
[12:42:02][I][ble_client:383]:  characteristic B42E4DCC-ADE7-11E4-89D3-123B93F75CBA, handle 0xd, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic B42E50D8-ADE7-11E4-89D3-123B93F75CBA, handle 0x10, properties 0x2c
[12:42:02][I][ble_client:383]:  characteristic B42E538A-ADE7-11E4-89D3-123B93F75CBA, handle 0x14, properties 0x10
[12:42:02][I][ble_client:170]: Service UUID: F000FFC0-0451-4000-B000-000000000000
[12:42:02][I][ble_client:171]:   start_handle: 0x17  end_handle: 0x23
[12:42:02][I][ble_client:383]:  characteristic F000FFC1-0451-4000-B000-000000000000, handle 0x19, properties 0x1c
[12:42:02][I][ble_client:383]:  characteristic F000FFC2-0451-4000-B000-000000000000, handle 0x1d, properties 0x1c
[12:42:02][I][ble_client:383]:  characteristic F000FFC5-0451-4000-B000-000000000000, handle 0x21, properties 0x14
[12:42:02][I][ble_client:170]: Service UUID: 0x180A
[12:42:02][I][ble_client:171]:   start_handle: 0x24  end_handle: 0xffff
[12:42:02][I][ble_client:383]:  characteristic 0x2A23, handle 0x26, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A24, handle 0x28, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A25, handle 0x2a, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A26, handle 0x2c, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A27, handle 0x2e, properties 0x2
[12:42:02][I][ble_client:383]:  characteristic 0x2A29, handle 0x30, properties 0x2
[12:42:02][W][airthings_wave_plus:030]: No sensor characteristic found at service B42E1C08-ADE7-11E4-89D3-123B93F75CBA char B42E2A68-ADE7-11E4-89D3-123B93F75CBA

janchrillesen, 2022-07-15:
The UUID's used and the structure for the Wave seems to be documented here - https://github.com/custom-components/sensor.airthings_wave/blob/master/custom_components/airthings_wave/airthings.py - I would still prefer an ESPhome based solution over the custom component, as my HomeAssistant server is not within bluetooth reach of my Airthings device

davidbb, 2022-07-27:
Here is a temporary workaround using ble_client sensor platform:

```
sensor:
  - platform: ble_client
    ble_client_id: airthings_radon
    update_interval: 5min 
    name: "Radon"
    icon: 'mdi:radioactive'
    service_uuid: 'B42E4A8E-ADE7-11E4-89D3-123B93F75CBA'
    characteristic_uuid: 'b42e4dcc-ade7-11e4-89d3-123b93f75cba'
    unit_of_measurement: "Bq/m³"
    lambda: |-
      uint16_t radon = x[4];
      radon += (x[5] << 8);
      return (float)radon;
  - platform: ble_client
    ble_client_id: airthings_radon
    update_interval: 5min 
    name: "Radon long term"
    icon: 'mdi:radioactive'
    service_uuid: 'B42E4A8E-ADE7-11E4-89D3-123B93F75CBA'
    characteristic_uuid: 'b42e4dcc-ade7-11e4-89d3-123b93f75cba'
    unit_of_measurement: "Bq/m³"
    lambda: |-
      uint16_t radon_lt = x[6];
      radon_lt += (x[7] << 8);
      return (float)radon_lt;

ble_client:
  - mac_address: XX:XX:XX:XX:XX:XX
    id: airthings_radon

esp32_ble_tracker:
```
The radon values are stored in the 5th and 6th bytes (current value), and 7th and 8th bytes (long term values), so the lambda on each just adds them up and returns a float.

janchrillesen, 2022-07-28:
Hi @davidbb - thank you very much. I can confirm that it works perfectly for reading the radon data.

bcochrane, 2022-08-22:
I also have an AirThings Wave (non-Plus model). As my first contribution to the ESPHome project, I will attempt to add support for this model.

bcochrane, 2022-08-31:
After digging into this a bit, I realized that there are two different versions of the Airthings Wave: 1st and 2nd generation.

Like the Wave Plus and Wave Mini, the Wave Gen2 packs values from all the sensors into a byte string available through a single Bluetooth LE characteristic, therefore it should be fairly straightforward to add Gen2 support following the pattern used for the Plus and Mini.

The Gen1, however, uses separate characteristics for each sensor, so it will require a different approach. I think it would be cleaner to add separate components for the Gen1 and Gen2 devices (perhaps named airthings_wave_gen1 and airthings_wave_gen2), rather than lump the functionality for both into one component, but I am definitely open to suggestions from other devs.

kevin-david, 2022-11-19:
Thanks @davidbb for providing your example. For anyone else who might want it, here's a complete example that also includes temperature and humidity from the Wave Radon, using US units: https://gist.github.com/kevin-david/f86a46147ef49b3c4d09205436adc7c2

anando-c, 2024-04-17:
@kevin-david I used the code from [here](https://github.com/kevin-david/home-assistant-shared/blob/main/esphome/esp32-wave-radon-proxy.yml) with my ESP32 device and I am unable to get sensor readings from my Wave (non-plus) Radon Gen2. Here is the log output from my ESP32 device - is there something that I am doing wrong? The Airthings app on my iPhone has no issues connecting to the device and grabbing all the data.

```
[15:26:09][D][esp32_ble_tracker:266]: Starting scan...
[15:26:20][D][esp-idf:000]: W (322646) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x8

[15:26:20][D][esp-idf:000]: W (322651) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8

[15:26:20][D][esp-idf:000]: W (322656) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x8

[15:26:20][D][esp-idf:000]: W (322658) BT_APPL: gattc_conn_cb: if=6 st=0 id=6 rsn=0x8

[15:26:20][D][esp-idf:000]: W (322661) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8

[15:26:20][D][esp32_ble_client:172]: [3] [F0:F8:F2:4F:5B:5B] ESP_GATTC_DISCONNECT_EVT, reason 8
[15:26:20][D][esp32_ble_client:110]: [3] [F0:F8:F2:4F:5B:5B] ESP_GATTC_CLOSE_EVT
[15:26:20][W][ble_sensor:037]: [Airthings 2900143582 Radon: Short Term] Disconnected!
[15:26:20][W][ble_sensor:037]: [Airthings 2900143582 Radon: Long Term] Disconnected!
[15:26:20][W][ble_sensor:037]: [Airthings 2900143582 Temperature] Disconnected!
[15:26:20][W][ble_sensor:037]: [Airthings 2900143582 Relative Humidity] Disconnected!
[15:26:21][D][esp32_ble_client:110]: [3] [F0:F8:F2:4F:5B:5B] Found device
[15:26:21][D][esp32_ble_tracker:661]: Found device F0:F8:F2:4F:5B:5B RSSI=-89
[15:26:21][D][esp32_ble_tracker:682]:   Address Type: PUBLIC
[15:26:21][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[15:26:21][I][esp32_ble_client:067]: [3] [F0:F8:F2:4F:5B:5B] 0x00 Attempting BLE connection
[15:26:22][D][esp32_ble_client:110]: [3] [F0:F8:F2:4F:5B:5B] ESP_GATTC_CONNECT_EVT
[15:26:22][D][esp32_ble_client:110]: [3] [F0:F8:F2:4F:5B:5B] ESP_GATTC_OPEN_EVT
[15:26:22][I][ble_sensor:031]: [Airthings 2900143582 Radon: Short Term] Connected successfully!
[15:26:22][I][ble_sensor:031]: [Airthings 2900143582 Radon: Long Term] Connected successfully!
[15:26:22][I][ble_sensor:031]: [Airthings 2900143582 Temperature] Connected successfully!
[15:26:22][I][ble_sensor:031]: [Airthings 2900143582 Relative Humidity] Connected successfully!
[15:26:22][D][esp32_ble_client:110]: [3] [F0:F8:F2:4F:5B:5B] ESP_GATTC_SEARCH_CMPL_EVT
[15:26:22][I][esp32_ble_client:227]: [3] [F0:F8:F2:4F:5B:5B] Connected
[15:26:22][W][ble_sensor:049]: No sensor characteristic found at service B42E4A8E-ADE7-11E4-89D3-123B93F75CBA char B42E4DCC-ADE7-11E4-89D3-123B93F75CBA
[15:26:22][W][ble_sensor:049]: No sensor characteristic found at service B42E4A8E-ADE7-11E4-89D3-123B93F75CBA char B42E4DCC-ADE7-11E4-89D3-123B93F75CBA
[15:26:22][W][ble_sensor:049]: No sensor characteristic found at service B42E4A8E-ADE7-11E4-89D3-123B93F75CBA char B42E4DCC-ADE7-11E4-89D3-123B93F75CBA
[15:26:22][W][ble_sensor:049]: No sensor characteristic found at service B42E4A8E-ADE7-11E4-89D3-123B93F75CBA char B42E4DCC-ADE7-11E4-89D3-123B93F75CBA
[15:26:22][W][component:232]: Component esp32_ble took a long time for an operation (81 ms).
[15:26:22][W][component:233]: Components should block for at most 30 ms.
```

kevin-david, 2024-04-27:
@anando-c if it's not supported out of the box - https://esphome.io/components/sensor/airthings_ble.html... you'll have to do some sniffing and experimentation: https://esphome.io/components/ble_client.html#setting-up-devices

Unfortunately it looks like the repo is gone that I sourced my info from. Maybe this will help? https://cc.bingj.com/cache.aspx?q=cached%3ahttps%3a%2f%2fztroop.github.io%2fwave-reader-utils%2fspecs%0d%0a&d=4822582936034875&mkt=en-US&setlang=en-US&w=z4x_yqndToKRDv_-Zwhta5rpxtWphkIP

t0ny-peng, 2024-10-23:
Hi @kevin-david huge thanks for sharing the code. May I ask if the following way of interpreting the bluetooth data is officially documented somewhere, or is it what you got from trial and error? Thanks!

```
      uint16_t radon_sta = x[4];
      radon_sta += (x[5] << 8);
      return (float)(radon_sta / 37.0f);
```

kevin-david, 2024-10-23:
@t0ny-peng I don't think it's officially documented, but the [source I used](https://github.com/esphome/feature-requests/issues/1795#issuecomment-1320749213) is gone. Lesson learned on archiving things.

The `/ 37.0f` is to convert from pCi/L from Bq/m3 - https://help.airthings.com/en/articles/3119854-radon-what-is-the-difference-between-pci-l-and-bq-m3

bcochrane, 2024-10-23:
@t0ny-peng In addition to the links provided by kevin-david, the [airthings-ble](https://github.com/Airthings/airthings-ble/tree/main) Python library may also provide some insight. This is used by the Airthings BLE integration in Home Assistant, which I'm successfully using with my Wave gen1.

I think the relevant parsing code is in the `_decode_wave` function in [airthings_ble/parser.py](https://github.com/Airthings/airthings-ble/blob/main/airthings_ble/parser.py).

kevin-david, 2024-10-23:
> @t0ny-peng In addition to the links provided by kevin-david, the [airthings-ble](https://github.com/Airthings/airthings-ble/tree/main) Python library may also provide some insight. This is used by [...]

Nice find. This would be good to cross-reference with my existing code to see if there's anything else we can pull off the device!

precurse, 2025-03-22:
> Here is a temporary workaround using ble_client sensor platform:
>
> ```
> sensor:
>   - platform: ble_client
>     ble_client_id: airthings_radon
>     update_interval: 5min
>     name: "Radon"
>     icon: [...]

Thanks for the template! That was a really good starting point. I improved it a bit on mine, so it extracts more data out of the BLE characteristic. It only queries BLE once, to help keep battery draw down on the Airthings sensor. 

```
sensor:
  - platform: ble_client
    ble_client_id: airthings_radon
    id: airthings_raw_ble
    update_interval: 5min 
    name: "BLE Raw"
    type: characteristic
    internal: True
    service_uuid: 'B42E4A8E-ADE7-11E4-89D3-123B93F75CBA'
    characteristic_uuid: 'b42e4dcc-ade7-11e4-89d3-123b93f75cba'
    lambda: |-
      uint8_t humidity = x[1];  // humidity
      uint8_t ambient_light = x[2]; // ambient
      // x[3] // unused
      uint16_t radon_st = x[4] | (x[5] << 8);       // short-term radon
      uint16_t radon_lt = x[6] | (x[7] << 8);       // long-term radon
      int16_t temperature_raw = x[8] | (x[9] << 8); // temperature
      bool occupancy = x[16] != 0;                  // occupancy
      // Calculate
      float light_level = (float)ambient_light;
      float hum_percent = (float)humidity;
      float temperature = temperature_raw / 100.0;
      // Update sensors
      id(ambient_light_sensor).publish_state(light_level);
      id(radon_sensor).publish_state(radon_st);
      id(radon_longterm_sensor).publish_state(radon_lt);
      id(temperature_sensor).publish_state(temperature);
      id(humidity_sensor).publish_state(hum_percent);
      id(occupancy_sensor).publish_state(occupancy);
      return {};
      
  - platform: template
    id: radon_sensor
    name: "Radon Short-Term"
    icon: 'mdi:radioactive'
    unit_of_measurement: "Bq/m³"
  - platform: template
    id: radon_longterm_sensor
    name: "Radon Long-Term"
    icon: 'mdi:radioactive'
    unit_of_measurement: "Bq/m³"
  - platform: template
    id: temperature_sensor
    icon: 'mdi:thermometer'
    name: "Temperature"
    unit_of_measurement: "°C"
  - platform: template
    id: humidity_sensor
    icon: "mdi:water-percent"
    name: "Humidity"
    unit_of_measurement: "%"
  - platform: template
    id: ambient_light_sensor
    name: "Ambient Light"
    unit_of_measurement: "lx"

binary_sensor:
  - platform: template
    id: occupancy_sensor
    icon: "mdi:home-outline"
    name: "Occupancy"
```

kevin-david, 2025-03-23:
>I ended up creating an external component that updates the UUIDs here https://github.com/precurse/esphome_airthings_wave_radon. It seems to be working for me. The awesome part is that this could probably be configured with a component flag, rather than having to write a whole new module.

@precurse It works great! Full example that works for me [here](https://github.com/kevin-david/home-assistant-shared/blob/bbec1b7dfe6ae00352964d51f08309d705a94bc6/esphome/esp32-wave-radon-proxy.yml#L36), including template sensors to add back US radon units.