esphome/feature-requests #1798: OTA update on wake up from sleep - "Persistant" OTA

luixal, 2022-07-17:
**Describe the problem you have/What new integration you would like**
I'm using esphome with devices that do long deep sleep times and small running times when they wake up. Applying and OTA update on these devices is kind of tricky but it's a solved problem by controlling the going to sleep manually:

```yaml
deep_sleep:
  # ...
  id: deep_sleep_1
mqtt:
  # ...
  on_message:
    - topic: livingroom/ota_mode
      payload: 'ON'
      then:
        - deep_sleep.prevent: deep_sleep_1
    - topic: livingroom/sleep_mode
      payload: 'ON'
      then:
        - deep_sleep.enter: deep_sleep_1
```

Thing is, probably you have many devices that wake up at different times and these way makes you handle all this "hey, stay awake" and going manually updating each one. This causes some problems: you keep devices running for quite long times (compared to how they usually behave) for updating all of them. Also, you have to update them manually once they are waked up.

**Please describe your use case for this integration and alternatives you've tried:**
Basically every sensor I got running on esphome: water tank level, ph- tank level, temp&humidity sensors... Right now, I'm using the existing solution and I'm keep up 10 temp&humidity sensors until all of them are up and then I update all of them one by one manually.

It would be great if I could edit yaml configs, save them and when selecting install, esphome handles the prevent sleep logic so all of them are updated the next time they wake up and continue working as usual, running and going to deep sleep.

peterk, 2023-08-05:
This is a great idea and I have the sam use case. A scheduled update would simplify a lot of things. To verify the update was installed it would be great if there was a way of displaying last update date. I guess it could be written by logging but time stamp of last update could maybe be a core property?

kpfleming, 2024-06-26:
The new [Managed Updates via HTTP Request](https://esphome.io/components/update/http_request) feature could be used to implement this.

luixal, 2024-08-12:
As @kpfleming points out, this new feature could cover this. Haven't tried it, but something like:

```yaml
esphome:
  on_boot:
    then:
      update:
        - platform: http_request
          name: Firmware Update
          source: http://example.com/manifest.json
```

or this, not sure:

```yaml
esphome:
  on_boot:
    then:
      - ota.http_request.flash:
        md5_url: http://example.com/firmware.md5
        url: https://example.com/firmware.ota.bin
```

I would only use this inside a local network (or VPN) as the best security you can get it is basic or including an API KEY in headers.

closed 2024-08-12 by luixal