esphome/feature-requests #1802: Add forced calibration to the SCD30

anoblet, 2022-07-18:
**Describe the problem you have/What new integration you would like**
Add forced calibration to the SCD30

**Please describe your use case for this integration and alternatives you've tried:**
Forced calibration is needed for indoor environments.

**Additional context**
This has already been implemented for SCD4X (https://esphome.io/components/sensor/scd4x.html#perform-forced-calibration-action)

Previous attempts:

* https://github.com/esphome/esphome/pull/1237
* https://github.com/esphome/esphome/pull/1402
* https://github.com/esphome/esphome/pull/1561

bartoszkaminski, 2022-09-29:
Any updates on it? I'm also looking forward for this functionality

bekriebel, 2022-10-12:
I'm making another attempt at this. I haven't opened a PR yet, but will after I do some more testing. If anyone else would like to try it:

https://github.com/bekriebel/esphome/tree/scd30_force_calibration_with_reference

```yaml
external_components:
  - source: github://bekriebel/esphome@scd30_force_calibration_with_reference
    components: [ scd30 ]
```

You can then call it as an action. Personally, I tied it to a button for now; this could easily be made into a Home Assistant service:

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: 38
      inverted: true
    name: "$friendly_name Button"
    on_press:
      then:
        - scd30.force_recalibration_with_reference:
            value: 419
```

WinstonCh, 2022-10-23:
This seems very promising! I am having issues with the external components, purely because of missing knowledge. I add your code example to the corresponding .yaml in ESPHome and get the following error:

`INFO Reading configuration /config/esphome/co2.yaml...
ERROR Unable to import component scd30.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/2fea77b5/esphome/components/scd30/sensor.py", line 6, in <module>
    from esphome.components import sensirion_common
ImportError: cannot import name 'sensirion_common' from 'esphome.components' (/esphome/esphome/components/__init__.py)
Failed config

sensor.scd30: [source /config/esphome/co2.yaml:47]
  
  Platform not found: 'sensor.scd30'.`

Is that anything you could help with? 

Cheers

bekriebel, 2022-10-24:
@WinstonCh I'm not quite sure what you need help with. If it helps, this is my sensor entry in my esphome config:

```yaml
substitutions:
  # Name the device and it's entities
  device_name: "test-air-quality"
  friendly_name: Test Air Quality

sensor:
  # Air quality sensor
  - platform: scd30
    id: sensor_scd30
    co2:
      name: "$friendly_name CO2"
      id: scd30_co2
      accuracy_decimals: 0
      filters:
        - filter_out: 0.0
        - filter_out: 40000.0
    temperature:
      name: "$friendly_name Temperature"
      accuracy_decimals: 2
    humidity:
      name: "$friendly_name Humidity"
      accuracy_decimals: 1
    automatic_self_calibration: false
    altitude_compensation: 100m
    update_interval: 60s
```

WinstonCh, 2022-10-24:
Thanks for your swift reply, @bekriebel, appreciated. 
I struggle with understanding how to call the forced recalibration, in particular around

```
binary_sensor:
  - platform: gpio
    pin:
      number: 38
      inverted: true
```

This'd be an excerpt of the log. 

```
[19:22:42][C][logger:278]:   Hardware UART: UART0
[19:22:42][C][i2c.arduino:038]: I2C Bus:
[19:22:42][C][i2c.arduino:039]:   SDA Pin: GPIO4
[19:22:42][C][i2c.arduino:040]:   SCL Pin: GPIO5
[19:22:42][C][i2c.arduino:041]:   Frequency: 50000 Hz
[19:22:42][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[19:22:42][I][i2c.arduino:054]: Results from i2c bus scan:
[19:22:42][I][i2c.arduino:060]: Found i2c device at address 0x61
[19:22:42][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'CO2 FRC'
[19:22:42][C][gpio.binary_sensor:016]:   Pin: GPIO5
[19:22:42][C][scd30:115]: scd30:
[19:22:42][C][scd30:116]:   Address: 0x61
[19:22:42][C][scd30:134]:   Altitude compensation: OFF
```

lucianlepadatu, 2022-11-28:
Hi @bekriebel
I've also tested this on a few sensors and the results look good.
Something that is not obvious and worth mentioning as others might get confused:
_The FRC method imposes a permanent update of the CO2 calibration curve which persists after repowering the sensor. The most recently used reference value is retained in volatile memory and can be read out with the command sequence given below. **After repowering the sensor, the command will return the standard reference value of 400 ppm**._
[https://sensirion.com/resource/user_guide/scd30/interface_description](https://sensirion.com/resource/user_guide/scd30/interface_description)

Would love to see your code merged :)

0x53A, 2022-12-01:
Hi @bekriebel thank you for the change!

I didn't have a physical button available, so I created a HA service you can call with a parameter:

```
api:
  services:
    - service: recalibrate_scdx_co2
      variables:
        actual_co2: int
      then:
        - scd30.force_recalibration_with_reference:
            value: !lambda 'return actual_co2;'
```

Works flawlessly.

bzeiss, 2022-12-11:
I get a linking error when I try to add this external component with the service on a raspberry pico device. It builds fine if I leave out the service (though I don't have a button). For an ESP32, it seems to build fine with a service.
`
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /data/air-test/.pioenvs/air-test/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJlEE15execute_serviceERKNS0_21ExecuteServiceRequestE':
main.cpp:(.text._ZN7esphome3api15UserServiceBaseIJlEE15execute_serviceERKNS0_21ExecuteServiceRequestE[_ZN7esphome3api15UserServiceBaseIJlEE15execute_serviceERKNS0_21ExecuteServiceRequestE]+0x216): undefined reference to `_ZN7esphome3api21get_execute_arg_valueIlEET_RKNS0_22ExecuteServiceArgumentE'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /data/air-test/.pioenvs/air-test/src/main.cpp.o: in function `_ZN7esphome3api15UserServiceBaseIJlEE28encode_list_service_responseEv':
main.cpp:(.text._ZN7esphome3api15UserServiceBaseIJlEE28encode_list_service_responseEv[_ZN7esphome3api15UserServiceBaseIJlEE28encode_list_service_responseEv]+0x2a): undefined reference to `_ZN7esphome3api19to_service_arg_typeIlEENS0_5enums14ServiceArgTypeEv'
collect2: error: ld returned 1 exit status
*** [/data/air-test/.pioenvs/air-test/firmware.elf] Error 1
`

nikolasgd, 2023-01-18:
Any help needed to get this merged? I would also love this functionality added.

ftrueck, 2023-01-29:
https://github.com/esphome/esphome/pull/4362

I took the changes and integrated them into a fresh fork. It seems the original fork is already too far behind the actual tree.
I also made a PR and updated the documentation for the module. Lets see when it will be merged into the dev branch.