esphome/feature-requests #1805: Pressure and temperature sensor JHM1200 chip

interkom, 2022-07-20:
**Describe the problem you have/What new integration you would like**
Please add a sensor:
Low-power IoT pressure sensor, 3.3V, I2C power supply, 0-1 MPa communication pressure sensor
https://aliexpress.ru/item/1005002685786637.html?gatewayAdapt=glo2rus&sku_id=12000021713435204&spm=a2g0s.12269583.0.0.53257059wZYXzZ

**Please describe your use case for this integration and alternatives you've tried:**
I found on the forums that this sensor has a JHM1200 chip
The documentation that the seller sent me is attached ( https://we.tl/t-DtHOi7bPba )
I'm not a programmer, after reading the documentation and searching the Internet for examples, I tried to add myself, but it didn't work out.
I tried with Arduino, I get a response from the sensor.

Sensor polling in Arduino
#include <Wire.h>

void setup() {
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
}

void loop()
{
  Wire.beginTransmission( 0x78);
  Wire.write( 0xAC);
  Wire.endTransmission();

  delay( 1000);

  int n = Wire.requestFrom( 0x78, 6);
  Serial.print( "Received ");
  Serial.print( n);
  Serial.print( " bytes: ");
  for( int i=0; i<n; i++)
  {
    Serial.print( Wire.read(), HEX);
    Serial.print( ", ");
  }
  Serial.println();

  delay( 1000);
}

Sensor response during the survey
Received 25 bytes: 4, 18, 8A, B9, 95, 5F

As far as I understand, here is a piece of code for decrypting the response data (from the documentation of the sensor seller)
The full code is in the documentation (file JHM1200.c)

#define CAL_L -125000  //0-1MPA
#define CAL_H 1125000 //0-1MPA

    BSP_IIC_Read(Device_Address, buffer, 6);//通过IIC读取数据字节
 
    //根据校准范围将返回的压力和温度值转换为实际值
    press_raw = ((ulong)buffer[1] << 16) | ((ulong)buffer[2] << 8) | buffer[3];
    temp_raw = ((ulong)buffer[4] << 8) | (buffer[5] << 0);
    press = (double)press_raw / 16777216;
    press = press * (CAL_H - CAL_L) + CAL_L;
    temp = (double)temp_raw / 65536;
    temp = temp * 19000 - 4000;
    my_temp = press;// 压力数值单位为PA

Please help me add a sensor.

**Additional context**

lemIY, 2023-03-14:
pressure.yaml
---------------------
esphome:
  name: "pressure"
  friendly_name: "pressure"
  includes:
    - pressure/custom_sensor.h
  libraries:
    - Wire    

esp8266:
  board: d1_mini

logger:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "New Fallback Hotspot"
    password: !secret esphome_fallback_password

captive_portal:

sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor->temperature_sensor, my_sensor->pressure_sensor};

  sensors:
  - name: "Return water temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1    
  - name: "Heating Water Pressure"
    unit_of_measurement: bar
    accuracy_decimals: 2

custom_sensor.h
-----------------------
#include "esphome.h"
#include <Wire.h>

#define CAL_L -125000  //0-1MPA
#define CAL_H 1125000 //0-1MPA   
#define DEVICE_ADDRESS (0x78)
 
void DelayMs(int ms) //8MHz/1MS
{
 delay(ms*1);
}

double my_press;
double my_temp;

uint8_t read8(byte reg) {
  uint8_t value;

  Wire.beginTransmission((uint8_t)DEVICE_ADDRESS);
  Wire.write((uint8_t)reg);

  if (0 == Wire.endTransmission(false)) { 
  Wire.requestFrom((uint8_t)DEVICE_ADDRESS, (byte)1);
  value = Wire.read();
  }
  else
    return 0;
  return value;
}

void write8(byte value) {

  Wire.beginTransmission((uint8_t)DEVICE_ADDRESS);
  Wire.write((uint8_t)value);
  Wire.endTransmission();
}
bool JHM1200_IsBusy()
{
 
  char status;
  status=read8(0xAC);
  status = (status >> 5) & 0x01;
  return status;
}
	
void JHM1200_get_cal()

{
  char buffer[6] = {0};
  ulong press_raw = 0;
  uint temp_raw = 0;
  double press = 0.0, temp = 0.0;
  buffer[0] = 0xAC;
  write8(0xAC);

  while (1)
  {
    if (JHM1200_IsBusy())
    {
      DelayMs(1);
    }
    else
      break;
  }

    Wire.requestFrom(DEVICE_ADDRESS, 6);
    int i=0;
    
    while (Wire.available()) {
      char c = Wire.read();
      buffer[i]=c;
       i++;
    }
	
    press_raw = ((ulong)buffer[1] << 16) | ((ulong)buffer[2] << 8) | buffer[3];
    temp_raw = ((ulong)buffer[4] << 8) | (buffer[5] << 0);
    press = (double)press_raw / 16777216; 
    press = press * (CAL_H - CAL_L) + CAL_L;
    temp = (double)temp_raw / 65536;
	my_press = press / 100000; // давление -> bar
    my_temp = 100 - (((double) temp * 19000 - 4550) / 100); // температура -> °C

}
//---------------------------------------------- in esphome -------------------------------- 

class MyCustomSensor : public PollingComponent, public Sensor {
 public:
  Sensor *temperature_sensor = new Sensor();
  Sensor *pressure_sensor = new Sensor();
  
  MyCustomSensor() : PollingComponent(15000) {}

  float get_setup_priority() const override { return esphome::setup_priority::BUS; }

  void setup() override {
    Wire.begin();
    Serial.begin(9600);
  }
  void loop() override {
    delay(100);
    JHM1200_get_cal();
    delay(300);
  }  

  void update() override {

    float temperature = my_temp;
    temperature_sensor->publish_state(my_temp);
    int pressure = my_press;
    pressure_sensor->publish_state(my_press);  
  }  

};

lemIY, 2023-03-14:
Used https://forum.arduino.cc/t/my-i2c-device-response-the-same-value-until-i-reconnect-the-power/878743
It turned out to forward to HA

pressure.yaml
---------------
esphome:
  name: "pressure"
  friendly_name: "pressure"
  includes:
    - pressure/custom_sensor.h
  libraries:
    - Wire    

esp8266:
  board: d1_mini

logger:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "New Fallback Hotspot"
    password: !secret esphome_fallback_password

captive_portal:

sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor->temperature_sensor, my_sensor->pressure_sensor};

  sensors:
  - name: "Return water temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1    
  - name: "Heating Water Pressure"
    unit_of_measurement: bar
    accuracy_decimals: 2

custom_sensor.h
-------------------

#include "esphome.h"
#include <Wire.h>

#define CAL_L -125000  //0-1MPA
#define CAL_H 1125000 //0-1MPA   
#define DEVICE_ADDRESS (0x78)
 
void DelayMs(int ms) //8MHz/1MS
{
 delay(ms*1);
}

double my_press;
double my_temp;

uint8_t read8(byte reg) {
  uint8_t value;

  Wire.beginTransmission((uint8_t)DEVICE_ADDRESS);
  Wire.write((uint8_t)reg);

  if (0 == Wire.endTransmission(false)) { 
  Wire.requestFrom((uint8_t)DEVICE_ADDRESS, (byte)1);
  value = Wire.read();
  }
  else
    return 0;
  return value;
}

void write8(byte value) {

  Wire.beginTransmission((uint8_t)DEVICE_ADDRESS);
  Wire.write((uint8_t)value);
  Wire.endTransmission();
}
bool JHM1200_IsBusy()
{
 
  char status;
  status=read8(0xAC);
  status = (status >> 5) & 0x01;
  return status;
}
	
void JHM1200_get_cal()

{
  char buffer[6] = {0};
  ulong press_raw = 0;
  uint temp_raw = 0;
  double press = 0.0, temp = 0.0;
  buffer[0] = 0xAC;
  write8(0xAC);

  while (1)
  {
    if (JHM1200_IsBusy())
    {
      DelayMs(1);
    }
    else
      break;
  }

    Wire.requestFrom(DEVICE_ADDRESS, 6);
    int i=0;
    
    while (Wire.available()) {
      char c = Wire.read();
      buffer[i]=c;
       i++;
    }
	
    press_raw = ((ulong)buffer[1] << 16) | ((ulong)buffer[2] << 8) | buffer[3];
    temp_raw = ((ulong)buffer[4] << 8) | (buffer[5] << 0);
    press = (double)press_raw / 16777216; 
    press = press * (CAL_H - CAL_L) + CAL_L;
    temp = (double)temp_raw / 65536;
	my_press = press / 100000; // давление -> bar
    my_temp = 100 - (((double) temp * 19000 - 4550) / 100); // температура -> °C

}
//---------------------------------------------- передача в esphome -------------------------------- 

class MyCustomSensor : public PollingComponent, public Sensor {
 public:
  Sensor *temperature_sensor = new Sensor();
  Sensor *pressure_sensor = new Sensor();
  
  MyCustomSensor() : PollingComponent(15000) {}

  float get_setup_priority() const override { return esphome::setup_priority::BUS; }

  void setup() override {
    Wire.begin();
    Serial.begin(9600);
  }
  void loop() override {
    delay(100);
    JHM1200_get_cal();
    delay(300);
  }  

  void update() override {

    float temperature = my_temp;
    temperature_sensor->publish_state(my_temp);
    int pressure = my_press;
    pressure_sensor->publish_state(my_press);  
  }  

};