esphome/feature-requests #1807: Level Bolt/Lock Support

nmajin, 2022-07-23:
There is no trivial way to get the Level Bolt/Lock into Home Assistant so is it possible to integrate it with esphome as a BLE device to get the battery, status, and to control the lock and unlock events?

Would like to simply track the battery status, lock/unlock status, and to control the lock/unlock events using Home Assistant integrated with esphome. The HomeKit Controller does not work with BLE devices like the Level Bolt/Lock and there is a workaround with Homebridge to setup a proxy switch but it is a bit cumbersome in my opinion.

There is no Level Home Assistant integration due to a lack of an API from Level.

nmajin, 2022-12-11:
Although Home Assistant Bluetooth Proxy now solves for getting the lock into Home Assistant, I would be curious to know if we could get additional data via esphome, for example, the battery percentage or state (app currently shows "Good") - Bluetooth Proxy does not provide any data other than toggling the lock.

ssieb, 2022-12-12:
You will need to find some reference for the BLE characteristics.