esphome/feature-requests #1808: Support for Inkbird TH1 Plus

tannisroot, 2022-07-29:
**Describe the problem you have/What new integration you would like**
I have an Inkbird TH1-Plus sensor (see https://aliexpress.com/item/1005003056767899.html), but unfortunately it doesn't seem to work with Esphome's ble integration. It is detected in the logs, however when I add it in config, the reported values are just all unknown no matter how long I wait. 

**Please describe your use case for this integration and alternatives you've tried:**
It does however work with custom Passive BLE Monitor Integration from HACS when i turn on active scan, so I assume the issue is with ESPHome integration, seeing how it's not listed anywhere as supported. But unfortunately that integration doesn't practically work for me due to limited range of signal.

**Additional context**
I am running ESPHome on a ESP32-WROOM-32 board.
This is how it's detected on scan:
```
[11:04:20][D][esp32_ble_tracker:726]: Found device 49:21:09:09:6B:D9 RSSI=-89
[11:04:20][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[11:04:20][D][esp32_ble_tracker:749]:   Name: 'sps'
```

Exact config I used:
```
  - platform: inkbird_ibsth1_mini
    mac_address: 49:21:09:09:6B:D9
    temperature:
      name: "Inkbird IBS-TH1 Temperature"
    external_temperature:
      name: "Inkburd IBS-TH1 External Temperature"
    humidity:
      name: "Inkbird IBS-TH1 Humidity"
    battery_level:
      name: "Inkbird IBS-TH1 Battery Level"
```
Let me know what information i can provide for you.

nagyrobi, 2022-07-29:
Please ask in our Discord channel for faster response.

Yonghui, 2022-07-29:
Please turn on very verbose logging. 
# Enable logging
logger:
  level: VERY_VERBOSE

Please post your log. It will be similar to the following:

[12:49:38][VV][esp32_ble_tracker:497]: Parse Result:
[12:49:38][VV][esp32_ble_tracker:514]: Address: 49:42:08:00:45:FE (PUBLIC)
[12:49:38][VV][esp32_ble_tracker:516]: RSSI: -82
[12:49:38][VV][esp32_ble_tracker:517]: Name: 'sps'
[12:49:38][VV][esp32_ble_tracker:525]: Ad Flag: 6
[12:49:38][VV][esp32_ble_tracker:528]: Service UUID: 0xFFF0
[12:49:38][VV][esp32_ble_tracker:531]: Manufacturer data: AB.10.00.D7.8B.64.06 (7)
[12:49:38][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.03.02.F0.FF.04.09.73.70.73.0A.FF.75.08.AB.10.00.D7.8B.64.06 (23)
[12:49:38][VV][xiaomi_lywsd03mmc:021]: parse_device(): unknown MAC address.
[12:49:38][VV][xiaomi_lywsd03mmc:021]: parse_device(): unknown MAC address.
[12:49:38][VV][xiaomi_mhoc401:021]: parse_device(): unknown MAC address.
[12:49:38][VV][inkbird_ibsth1_mini:059]: parse_device(): unexpected data
[12:49:38][D][esp32_ble_tracker:723]: Found device 49:42:08:00:45:FE RSSI=-82
[12:49:38][D][esp32_ble_tracker:744]: Address Type: PUBLIC
[12:49:38][D][esp32_ble_tracker:746]: Name: 'sps'

tannisroot, 2022-07-29:
Sure, here, without the integration in config
[logs_esp32-board_logs.txt](https://github.com/esphome/feature-requests/files/9223332/logs_esp32-board_logs.txt)

tannisroot, 2022-07-29:
With integration in config
[logs_esp32-board_logs (2).txt](https://github.com/esphome/feature-requests/files/9223715/logs_esp32-board_logs.2.txt)
```
[01:41:37][VV][esp32_ble_tracker:500]: Parse Result:
[01:41:37][VV][esp32_ble_tracker:517]:   Address: 49:21:09:09:6B:D9 (PUBLIC)
[01:41:37][VV][esp32_ble_tracker:519]:   RSSI: -76
[01:41:37][VV][esp32_ble_tracker:520]:   Name: 'sps'
[01:41:37][VV][esp32_ble_tracker:528]:   Ad Flag: 6
[01:41:37][VV][esp32_ble_tracker:531]:   Service UUID: 0xFFF0
[01:41:37][VV][esp32_ble_tracker:534]:   Manufacturer data: F8.11.01.59.0A.46.06 (7)
[01:41:37][VV][esp32_ble_tracker:550]: Adv data: 02.01.06.03.02.F0.FF.04.09.73.70.73.0A.FF.FE.09.F8.11.01.59.0A.46.06 (23)
[01:41:37][VV][pvvx_mithermometer:021]: parse_device(): unknown MAC address.
[01:41:37][VV][xiaomi_miscale:019]: parse_device(): unknown MAC address.
[01:41:37][VV][inkbird_ibsth1_mini:059]: parse_device(): unexpected data
```

Yonghui, 2022-07-30:
Please look for a work around at #1586 
https://github.com/esphome/feature-requests/issues/1586

tannisroot, 2022-07-30:
That actually worked, thanks!

tannisroot, 2022-07-31:
While https://github.com/esphome/esphome/pull/3664 hasn't been merged yet and a new release is yet to be made, if you have the same type of device and you need to get it working in ESPHome, add this to your config:
```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: patch-2
    components: [ inkbird_ibsth1_mini ]
```
Be aware that this is a temporary solution and it will probably break in the future!

closed 2022-08-01 by jesserockz

ssieb, 2022-08-01:
It's merged now, so you can use the esphome dev branch instead or:
```yaml
external_components:
  - source: github://esphome/esphome@dev
    components: [ inkbird_ibsth1_mini ]
```