esphome/feature-requests #1810: Add 3g / 4g txt support uisng SIM7600 or similar

andypnz, 2022-07-30:
**Describe the problem you have/What new integration you would like**

Hi. I see there is a ESPHome integration for the SIM800L to send SMS txt messages..... Unfortunately the SIM800L uses the 2G network, which is being removed..... Is there any support for the SIM7600 or similar board to use the 3G / 4G network to send / receive txt messages using ESPHome running on a Wemos D1 mini or ESP32, etc ?

**Please describe your use case for this integration and alternatives you've tried:**

As above

**Additional context**

nagyrobi, 2022-07-31:
As far as I can see in most parts of the world 2G stays as it's based on plain GSM and used by older IoT, not planned to be removed any time soon. 3G is being renoved, and 4G is still in active operation as even 5G relies on it as NSA network.

Where are you located?

andypnz, 2022-07-31:
I am in New Zealand. Some cellular providers here have already shut down their 2G networks here..... Others plan to do the same by 2025 despite potentially millions of devices still using 2G..... So looking for a 3G / 4G solution.....

And I would like to send the sms direct from a ESPHome device, not something more central (like Home Assistant).

Thank you for your thoughts....

nagyrobi, 2022-07-31:
Don't invest on 3G as it will be shut down soon. You need a 4G/5G device on long term.

But given the super low cost of SIM800L it still might be worth using it until 2025 as it's supported at the moment by ESPHome.

andypnz, 2022-07-31:
Agree..... and I think the SIM7600 fits the bill..... hoping for a solution to use 4G/5G now :-)

andypnz, 2022-08-07:
Unfortunately the 800L is not working for me here - and confirmed 2g is already not supported on my chosen network. So I do need a 3G/4G/5G solution - I have ordered a SIM7600 - looking on GitHub I wonder if the 800L code code be modified to work with the 7600?

Happy to do some testing if anyone is interested (and has time) in doing that? My 7600 should be here in a week or so…..

bobzer, 2023-07-11:
> 

Did you had some success ?

robdejonge, 2023-10-11:
+1

I’m in Thailand and 2G is long gone here.

garethm79, 2023-11-03:
Has anybody tried actually using a SIM7000/7600/7670 etc using the SIM800 component? They all have UART and use the same AT command set so it might just work? I've just ordered a LillyGo A7670E module and ESP32 "motherboard" and will have a play when it arrives. It might just need some minor fiddling to get it to work.

NB: The 7670 is worth considering if you only need 4G/LTE and there's a version for your region. The 7600 supports "old" 3G comms and is physically larger and much more expensive. Note that the SIM7000 is LTE Cat-M/NB only, which might need a special IoT SIM plan rather than a regular consumer SIM, and might not even be supported on some networks.

randybb, 2023-11-03:
Yes, it works as these basic AT commands are the same. Tested with sim7600E.

chino-lu, 2023-11-09:
@andypnz please send me your mail. I have adapted the code that it should be working in case no GSM is available.
@randybb I can confirm the SIM800L code is working when GSM is available. If not, it will not be working since the code things the registration was not successful. I have modified slightly and looking for some testers

qnixAVA, 2023-12-28:
@chino-lu Hello, If it still possible, I would like to try your code for using SIM7600 by GSM/LTE with my HA server

danielionutmuresan, 2024-01-21:
@chino-lu I would like to test it as well of your offer is still standing. Thank you.

chino-lu, 2024-01-22:
@qnixAVA sorry, hadn't seen your message here. I have uploaded the code to https://github.com/chino-lu/sim7600

pacmac, 2024-02-29:
@qnixAVA - sorry for my lack of knowledge, but how would I go about using that code ?

qnixAVA, 2024-02-29:
Hello @pacmac, I don't know yet, I was thinking about another problem concerning my project. Now it's solved, I can start thinking about how to use the code of chino-lu on esphome.

chino-lu, 2024-02-29:
you only need it to place it in your config folder
config
- esphome
- - include
- - -SIM7600

in your esphome yaml you define
esphome:
  includes:
    - SIM7600/sim7600.h

pacmac, 2024-02-29:
Thanks so much, I have included it and it compiles with no errors, but how do I use it:

```
sim7600:
  on_sms_received:
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]
```

Throws error component not found

pacmac, 2024-02-29:
Have tried to figure it out, but compile fails with `Component not found: sim7600.`:

```
esphome:
  includes:
    - sim7600/sim7600.h

custom_component:
  - lambda: |-
      auto sim7600 = new sim7600();
      App.register_component(sim7600);
      return {sim7600};

sim7600:
  on_sms_received:
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]
```

chino-lu, 2024-02-29:
just checked again, here is my code

`esphome:
  name: kc868-aio
  friendly_name: KC868-AIO

external_components:
  - source:
      type: local
      path: include
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

ota:
  password: "xxxx" 

wifi:
  ssid: "xxxx"
  password: "xxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kc868-Aio Fallback Hotspot"
    password: "xxxxx"

captive_portal:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"
  services:
    - service: send_sms
      variables:
        recipient: string
        message: string
      then:
        - sim7600.send_sms:
            recipient: !lambda 'return recipient;'
            message: !lambda 'return message;'
    - service: dial
      variables:
        recipient: string
      then:
        - sim7600.dial:
            recipient: !lambda 'return recipient;'
    - service: connect
      then:
        - sim7600.connect
    - service: disconnect
      then:
        - sim7600.disconnect
    - service: send_ussd
      variables:
        ussdCode: string
      then:
        - sim7600.send_ussd:
            ussd: !lambda 'return ussdCode;'

text_sensor:
  - platform: template
    id: sms_sender
    name: "Sms Sender"
  - platform: template
    id: sms_message
    name: "Sms Message"
  - platform: template
    id: caller_id_text_sensor
    name: "Caller ID"
  - platform: template
    id: ussd_message
    name: "Ussd Code"

uart:
  tx_pin: GPIO13
  rx_pin: GPIO15
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);`

pacmac, 2024-02-29:
thanks so much that worked :-)

pacmac, 2024-03-01:
Well it seems that I can receive SMS but I cannot send and are not sure where to start debugging:

```
[10:00:05][D][uart_debug:158]: >>> "\x1AAT+CSCS=\"GSM\"\r\n"
[10:00:05][D][uart_debug:158]: <<< "\r\nOK\r\n"
[10:00:05][D][uart_debug:158]: >>> "AT+CMGS=\"+XXXXXXXXXX\"\r\n"
[10:00:05][I][sim7600:343]: Sending to +XXXXXXXXXX message: 'Pong'
[10:00:05][D][uart_debug:158]: <<< "\r\n>"
[10:00:05][D][uart_debug:158]: >>> "Pong\x1A"
[10:00:05][D][uart_debug:158]: <<< " \r\n"
[10:00:13][D][uart_debug:158]: <<< "\r\n+CMS ERROR: Unknown error\r\n"
```

Any suggestions ?

qnixAVA, 2024-03-01:
Hello there, I wanted to know if with that code I will be able to send data from sensor to my home assistant ?
I was looking for doing this by 4G/5G, but idk if it's possible to do this by sms ?

randybb, 2024-03-01:
@qnixAVA it will not - esphome can use only wifi or ethernet, data transfer via HSDPA/LTE is not implemented. Anyway, these SIM7600 modules have native http/https and mqtt build-in, so this could be one of the possible options. We just need to find a volunteer who will implement it :)

@chino-lu btw, maybe the network registration type could be configurable - AT+CREG for 3g/4g, AT+CGREG for GPRS and AT+CEREG for LTE only then we could have one universal component for all SIMxxxx modules.

qnixAVA, 2024-03-01:
@randybb okay.. thanks. Maybe it will be possible by using https webhook ? In this case I will need to establish connection of my SIM7600 first, then add webhook to my home assistant with esphome API  ?

qnixAVA, 2024-03-01:
@randybb Yes thank's buddy. I was thinking about since the start of my project but I don't have enough programming knowledge to adapt the arduino code to yaml of esphome :/ .

chino-lu, 2024-03-01:
> CSCS

could you try 00 instead of +?

trungnvdng, 2024-03-05:
I have this problem when compiling the code in esphome:
[image]

chino-lu, 2024-03-05:
please share you folder structure in ESPHome

trungnvdng, 2024-03-05:
The structure is:
ESPHome/  sim7600/    sim7600.h

chino-lu, 2024-03-06:
hm not sure how you are compiling, because this structure doesn't look like ESPHome in HomeAssistant

qnixAVA, 2024-03-24:
Hi guys, I'm starting to get interested in the 4G connection part of my project, I have an A7670E (quite similar to the sim7600 I think). 
I'm not yet very comfortable with custom components, and I'd like to know if you could help me set up a 4G connection when I start up my esp32?

The ultimate goal is for my esp32 to transmit data from its sensors to my home assistant server located 5km from my esp.

Thanks in advance.

Translated with DeepL.com (free version)

ChadMatsalla, 2024-05-03:
I'm an experienced ESPHome user. I recently got a Lilygo A7670G for monitoring bee hives in a remote location. I'm sending MQTT messages to a cloud instance. I then have a subscriber in my home that republishes the MQTT messages to my Homeassistant. It's slightly kludgy but better than opening ports to allow the device to send MQTT messages directly to Homeassistant. If anybody wants me to test code on this device I'm willing to do that.

ChadMatsalla, 2024-05-03:
I forgot to mention that I am NOT using ESPHome on this device at this time but I'm willing to test any code intended for the A7670G modem.

udiNur, 2024-05-06:
> Reference in new iss

Can you provide the code please?

ChadMatsalla, 2024-05-06:
I'm not sure what you mean by "Can you provide the code". It is *not* esphome code. It's code found here:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main

I have no intention of converting this to a component but if someone wants to do that I'd be willing to test it.

iBobik, 2025-04-02:
Here in Czechia GSM will be unavailable in 2 years, so we does not want to invest in 2G-only HW.

Anyone who tested Esphome SIM800 integration with some modern chip, would you consider [creating PR on the docs](https://github.com/esphome/esphome-docs/blob/current/components/sim800l.rst)? Add „Compatibility“ section and mention your HW and what features your tested.

deanfourie1, 2025-06-21:
Will this work with the LilyGo T-SIM7600G-H

closed 2026-10-23 by bdraco