esphome/feature-requests #1815: work around for the 1000steps/second cap on steppers

gammeflamme, 2022-08-03:
**Describe the problem you have/What new integration you would like**

When using a high gear ratio stepper motor such as the 28BYJ-48 with an tmc2209 stepper driver(minimum microstepping 1/8)
the 1000steps/second cap of the stepper component makes it impossible to rotate the stepper at a resonable pase.

**Please describe your use case for this integration and alternatives you've tried:**

I'm trying to use this motor driver combo for a quiet smart blind setup.

**Additional context**

nagyrobi, 2022-08-03:
Please use our Discord channel for support instructions.

gammeflamme, 2022-08-03:
> Please use our Discord channel for support instructions.

I did and i got the answer that the component would need to be changed.
[image]
d

nagyrobi, 2022-08-03:
Great! Thanks for the feedback.

iqubik, 2024-01-17:
need new esp32 stepper component