esphome/feature-requests #1818: Please add support for DART  WZ-S sensor

yhchen934, 2022-08-05:
**Describe the problem you have/What new integration you would like**

I have a DART sensor WZ-S, but I found that ESPHOME currently does not support it.
Please add support for DART WZ-S sensor.

**Please describe your use case for this integration and alternatives you've tried:**

DART WZ-S sensor is formaldehyde sensor, it can provide stable and good air quality monitoring, and there is currently no solution for connecting to ESPHOME.

**Additional context**

https://www.dart-sensors.com/products/formaldehyde/wz-s-formaldehyde-module/

nickrout, 2022-08-06:
It is serial, so it may be easy to get going via a custom uart sensor https://www.dart-sensors.com/wp-content/uploads/2017/07/formaldehyde_module_operation-manual.pdf

Qigaoo, 2022-08-14:
Please add support for DART WZ-S sensor.

Qigaoo, 2022-08-14:
> It is serial, so it may be easy to get going via a custom uart sensor https://www.dart-sensors.com/wp-content/uploads/2017/07/formaldehyde_module_operation-manual.pdf

If multiple such sensors are used, such as senseair S8 Co_ 2 sensor, pms7003 and dart formaldehyde sensors, it is difficult for most people to customize UART

yhchen934, 2022-08-14:
As a general user, it is quite difficult to handle this requirement by yourself. Please help us.

I saw a topic about dart and ESPHOME in the Chinese discussion forum, but it must be a user-defined function library, and it can only be used in older version of  ESPHOME.

I upload it for your reference.
[dart.zip](https://github.com/esphome/feature-requests/files/9334223/dart.zip)

yukunyi, 2023-07-05:
Request support too!

kenneth104, 2024-04-05:
Request support too!

gaodaxiaya, 2024-05-29:
please

alcadaa323, 2024-08-12:
Request support too!

latel, 2024-12-06:
Request support too!