esphome/feature-requests #1819: Add support for Hughes Power Watchdog Surge Protector

spbrogan, 2022-08-06:
**Describe the problem you have/What new integration you would like**

Add integration that provides sensors for voltage, Current, and Power usage (Line 1 and Line 2). 
Add support for error reporting when the surge protector reports power issue.
Add support for cumulative power usage reporting.  

**Please describe your use case for this integration and alternatives you've tried:**

Tried it with lambdas but because it uses vendor defined characteristic in a vendor defined message format split across multiple transmissions it seems overly difficult to do without custom code integration.  

**Additional context**

Product: https://hughesautoformers.com/product/pwd50/

Numerous emails have been sent to Hughes and they were unable to provide documentation or support for integrating the device.  There is an existing hughes mobile app but those details were not shared.

spbrogan, 2022-08-06:
Start of an integration is here: https://github.com/spbrogan/esphome/commits/dev

Will open code PR once more cleanup, tuning, and validation is done.  Help is always welcome.   To use in your own esphome instance here is how I have included it on a system running esphome dev docker container.  

``` yaml

external_components:
  - source:
      type: git
      url: https://github.com/spbrogan/esphome
      ref: dev
    components: [ hughes_power_watchdog ]
    refresh: 1s

ble_client:
  - mac_address: 34:14:b5:3c:93:92
    id: power_watchdog

sensor:
  - platform: hughes_power_watchdog
    ble_client_id: power_watchdog
    voltage_line_1:
      name: "Watchdog Voltage Line 1"
    current_line_1:
      name: "Watchdog Current Line 1"
    power_line_1:
      name: "Watchdog Power Line 1"
    voltage_line_2:
      name: "Watchdog Voltage Line 2"
    current_line_2:
      name: "Watchdog Current Line 2"
    power_line_2:
      name: "Watchdog Power Line 2"
    total_power:
      name: "Watchdog Cumulative Power Usage"
```

spbrogan, 2022-08-06:
This is the actual product being tested by the integration.  https://hughesautoformers.com/product/pwd50-epo/
The link in the feature request is a slightly different model.  Will need community support for testing with the various Power Watchdog products that are very similar.

tango2590, 2022-08-12:
This is a **highly** requested integration in the Smart RV community, since it's one of only a few products that can track our energy usage. Can confirm this is working on a 50amp hardwired version of the unit.

Additional information: A few people had reached out directly to the manufacturer, Hughes, to add integration themselves. While the manufacturer was all for the idea, they stated it was out of their hands as they contracted a 3rd party company to handle the app and bluetooth portion of the devices. No response from the 3rd party company.