esphome/feature-requests #1820: Add Pages support to ST7920 display

Wummeke, 2022-08-07:
**Describe the problem you have/What new integration you would like**

I salvaged a displayunit from an Ender3, it contains a rotary encoder and a ST7920 display. The display works fine in "Lamda-mode" but I would like to use pages, to I can scroll t hrough pages with the rotary encoder.
Although the documentation says it does supports pages, the display remains blank after uploading a sketch with pages.

**Please describe your use case for this integration and alternatives you've tried:**

This works fine:
```
display:
  - platform: st7920
    id: the_display
    cs_pin:
      number: 19
      inverted: true
    width: 128
    height: 64
    update_interval: 10s
    lambda: |-
      it.print(1, 32, id(lucon20), "This is LAMDA!!!11");
```
But I'd like this to work too:
```
display:
  - platform: st7920
    id: the_display
    cs_pin:
      number: 19
      inverted: true
    width: 128
    height: 64
    update_interval: 10s
    pages:
      - id: page1
        lambda: |-
          it.print(1, 32, id(lucon20), "This is page 1!");
      - id: page2
        lambda: |-
          it.print(1, 32, id(lucon20), "This is page 2!");
```

**Additional context**

nagyrobi, 2022-08-07:
See this, as a workaround: https://github.com/esphome/feature-requests/issues/1480#issuecomment-1175133578

nagyrobi, 2023-04-12:
See docs for solution: https://esphome.io/cookbook/lambda_magic.html#display-pages-alternative

closed 2023-04-12 by nagyrobi