esphome/feature-requests #1822: ESP32 presence detection with ESP-CSI

pluskal, 2022-08-08:
**Describe the problem you have/What new integration you would like**

A new way of human detection was presented https://github.com/espressif/esp-csi It would be interesting to use this as a sensor for presence detection to overcome the problem with PIRs and stationary objects.

**Please describe your use case for this integration and alternatives you've tried:**

N/A

**Additional context**

- https://stevenmhernandez.github.io/ESP32-CSI-Tool/
- https://www.youtube.com/watch?v=tFxKUzEDSdw

pat1300, 2022-08-08:
https://www.youtube.com/watch?v=Viqvx7hMMJs

evelant, 2024-07-01:
Has anybody made any progress in experimenting with this? It's really interesting stuff but I currently lack the background to attempt an implementation (and lack the time right now to acquire that math, radio, signal processing, and C background).