esphome/feature-requests #1826: Web Server Component - show unit_of_measurement on template number

MagMeister, 2022-08-10:
Meanwhile, unit_of_measurement is shown in Home Assistant after a [feature request](https://github.com/esphome/feature-requests/issues/1366), but not yet in the state of the ESPHome web server.

RoboMagus, 2022-10-28:
I cannot reproduce this issue. Is it still valid or can it be closed?

Config snippet and UI below:

```yaml
number:
- platform: template
  icon: "mdi:fan-speed-1"
  name: "Template Num"
  entity_category: config
  optimistic: true
  initial_value: 30
  unit_of_measurement: "%"
  min_value: 10
  max_value: 45
  step: 1
```
[image]

MagMeister, 2022-10-28:
I confirm that it has been solved. I'm running version 2022.10.1.

closed 2022-10-28 by MagMeister