esphome/feature-requests #1840: Option to prepend device name to all entity names at runtime

cybershoe, 2022-08-28:
**Describe the problem you have/What new integration you would like**

A configuration option to prepend all entity IDs exposed to HA with the effective device name (i.e. if `name_add_mac_suffix: true`, entity IDs would be prepended with the suffixed device name.)

**Please describe your use case for this integration and alternatives you've tried:**

I want to flash a number of custom devices with the same firmware binary, using `name_add_mac_suffix` to give them unique names that can be discovered by HA. I then want to be able to change the device name in HA to a more descriptive one, and also have the entity IDs (in HA, not ESPHome) match the new device, to make it easier to know which entity is associated with which device.

Currently, because the device name isn't known until runtime, I cannot make the entity names contain the device name as a substring, and therefore I must manually change each entity ID in HA for every device I discover.

**Additional context**

The HA device renaming process assumes that the device name is a substring of its entity IDs. This allows the entity IDs to be automatically updated when the device name is changed in HA. If the device name is fixed in the ESPHome configuration, then the entity names can be manually (or through substitutions) made to contain that substring, but if the device name is dynamic (appended MAC suffix), there is no way to have the entity names contain the entire device name (as the substitutions are done at compile time, not runtime).

cybershoe, 2022-08-28:
It would be nice to allow the suffixed device name to be used as the captive portal SSID as well, so that devices could be shipped with the captive portal enabled, allowing multiple devices to be onboarded without SSID conflicts.

nagyrobi, 2022-08-28:
Also see https://github.com/esphome/feature-requests/issues/1741.

dcousens, 2022-08-28:
A lambda would probably be easiest for this

cybershoe, 2022-08-28:
> A lambda would probably be easiest for this

The `name:` option is not templateable.