esphome/feature-requests #1844: Sprinkler Controller - support selecting weekdays, sunrise offset and forcasted temperature.

bogdanr, 2022-08-30:
The best option for irrigating the lawn is to do deep and less frequent irrgation sessions, in the early morning.

So, as an example, instead of wattering for 10 minutes every day it's better to watter for 20 minutes every two days. This way the watter reaches deeper in the soil and facilitates deep root growth. After the grass develops very deep roots it could be enough to watter 2-3 times a week. This is also nice because the top of the soil would not be very moist and there would be less harmfull insects in the first few centimeters of soil where the grass would be more vulnerable.

Also, it's good to watter in the early morning so that during the day the top layer of soil gets a chance to evaporate most of the watter. This minimizes moss formation. It would be nice to have a wattering schedule based on sunrise +/- some configurable offset.

Also, based on the forcasted temperature it would be nice to be able to modify the time the sprinklers run. Eg, if temperature is over 30°C run for extra 5 minutes, if the temperature is under 20°C run 5 minutes less than normal.

nagyrobi, 2022-09-19:
@kbx81

kbx81, 2022-09-19:
Maybe your first couple paragraphs are something we could add to the documentation?

> It would be nice to have a wattering schedule based on sunrise +/- some configurable offset.

I think this (as well as picking days of the week, etc.) would be achievable on-device if we had some form of a scheduling/cron system in ESPHome but...that doesn't seem to exist yet. It wouldn't make sense to build that into the sprinkler controller itself as many other components would expect to be able to use it. The topic of an ESPHome "scheduler" has surfaced repeatedly. We will get there...

> Also, based on the forcasted temperature it would be nice to be able to modify the time the sprinklers run. Eg, if temperature is over 30°C run for extra 5 minutes, if the temperature is under 20°C run 5 minutes less than normal.

Between the multiplier value, on-the-fly adjustment of run durations and the queue, this is already very achievable with a little automation (either in HA or potentially on the ESP itself, depending on your specific setup). That said, do you have a proposal to describe exactly how this would be implemented with respect to the yaml configuration? For example:

 - How would you expect to integrate the weather? At very least I expect you'd need to consider both temperature and conditions (rainy, sunny, cloudy, snowy, etc.). The ESP would need some consistent way to retrieve the weather to support this. Maybe via a [Home Assistant sensor](https://next.esphome.io/components/sensor/homeassistant.html)?
 - If other types of sensors (soil moisture/depth, for example) were to be used, would you expect to provide a sensor for each zone? What if you wanted to use multiple sensors?
 - What would your expectation be in terms of how the aforementioned points of data would affect the scheduling for any given valve/zone/the system as a whole?
   - How would you describe this desired behavior to the sprinkler controller in the configuration?
   - Would behavior be different for each valve/zone? Consider (for example) that one side of my property receives full sunlight all day while the other is largely shaded.
 - What should the controller do if it can't determine the weather conditions and/or soil moisture at any given time?

From what I've seen so far, most folks are tackling the integration of other sensors and weather conditions with automation(s). The `sprinkler` component itself just manages switching the valves on and off as required when it's triggered to do so.

nagyrobi, 2022-09-19:
For weather-based watering, I use a controller from https://opensprinkler.com/ (it's open source, by the way, but I opted to buy the ready-made hardware to sponsor the work of the author.

Weather-based adjustment is not something that can be easily calculated, there are algorithms for this (eg.  Zimmerman method, Evapotranspiration method etc.). The project author maintains a server which does these calculations and pushes down the results to the controllers every hour.

> weather-based adjustment first uses your location to query Wunderground.com for real-time weather data (available worldwide). It then calculates a % Watering value which is applied to all station water times. The percentage value is calculated based on your local temperature, humidity, and precipitation (rainfall), and varies between 0% to 250%. The value is updated every hours. For example, low temperature, high humidity, and/or considerable rainfall all lead to reduced watering, thus saving water usage.

See their [documentation](https://openthings.freshdesk.com/support/solutions/articles/5000823370-using-weather-adjustments) page for how they do this in detail.

Zebble, 2022-09-24:
> Weather-based adjustment is not something that can be easily calculated, there are algorithms for this (eg. Zimmerman method, Evapotranspiration method etc.) which need quite a lot of rather accurate input data (like temperature, humidity, precipitation, geolocation, altitude, solar radiation...). The project author maintains a server which does these calculations and pushes down the results to the controllers every hour.

I've been playing with this as well and will likely use the values that HASmartIrrigation provides to create a multipler that can be passed to the Sprinkler component (https://github.com/jeroenterheerdt/HAsmartirrigation).  Irrigation Unlimited (https://github.com/rgc99/irrigation_unlimited) uses HASmartIrrigation this way.

The multiplier on the ESPHome component can only go down to 0.1 so it would be nice if this could go all the way down to 0 when HASmartIrrigation decides we shouldn't be watering that day.

nagyrobi, 2022-09-24:
> The multiplier on the ESPHome component can only go down to 0.1 so it would be nice if this could go all the way down to 0 when HASmartIrrigation decides we shouldn't be watering that day.

@kbx81 this is something to consider!

kbx81, 2022-09-24:
> The multiplier on the ESPHome component can only go down to 0.1

This is not correct -- the only requirement is that it must just be [greater than zero](https://github.com/esphome/esphome/blob/6ef93452f5fa707d1023e498b7d3ee6579850411/esphome/components/sprinkler/sprinkler.cpp#L502-L508).

In terms of setting it _to zero_ -- I see two potential issues with this:

 - It creates a point of confusion in terms of behavior if the controller is ever used manually. For example, if the multiplier is set to zero, and you go to manually to start a zone to test/inspect it, what happens? Nothing. Unless you _happen_ to be looking at the logs when this situation arises, I guarantee you (or _someone_) is going to be standing there wondering why it won't switch on.
 -  Minimum run times are dictated by some of the overlap/delay settings. When you set an extremely tiny multiplier, if this results in run times shorter than any of these delay/overlap intervals, the value gets adjusted to still run for an interval one second greater than any configured delay/overlap. This is a limitation implemented due to the way the valve scheduler works.

The minimum run duration is easy to address -- if the multiplier is zero, it just does not start. However, again, I think this introduces confusing behavior.

That said, if the add-on/integration you are using computes a value of zero for the multiplier and you already have an automation that feeds this data into the `sprinkler` component, couldn't you just tweak your automation to not start the sprinkler controller when the computed multiplier value is zero?

nagyrobi, 2022-09-25:
Yeah, we could even use a template output to detect that in between the HA integration and sprinkler integration in ESPHome.