esphome/feature-requests #1845: Add restore_value for number component

IainPHay, 2022-08-31:
**Describe the problem you have/What new integration you would like**
Currently number components do not have a restore_value (unlike the template number). This means that after a reboot an input number value is lost and has to be reset. e.g. when using FSRs and ESPHome as a bed occupancy sensor as per 
https://community.home-assistant.io/t/fsr-the-best-bed-occupancy-sensor/365795/143

**Please describe your use case for this integration and alternatives you've tried:**
A current work around is to pull a value from home assistant instead where an input number is set, iit would be neater if it was contained within the ESPHome code on the device as would allow local activities to continue if HA is down,.

**Additional context**

nagyrobi, 2022-08-31:
> A current work around is to pull a value from home assistant instead

Please detail this workaround.

IainPHay, 2022-08-31:
Instead of using:

number:
  - platform: template
    name: "Bed Trigger Level"
    id: "bed_trigger_level"
    optimistic: true
    min_value: 0
    max_value: 3.5
    step: 0.05
    icon: mdi:arrow-collapse-down
    unit_of_measurement: V 

You can pull a value set in homeassistant using:

sensor 
  - platform: homeassistant
    name: "Bed Trigger Level"
    id: "bed_trigger_level"
    entity_id: input_number.bed_trigger_level
  
With an input number set in homeassistant

This value is used in:

binary_sensor:
  - platform: template
    name: "Bed Occupied"
    device_class: occupancy
    id: occupancy
    lambda: |-
      if (id(bed_sensor).state < id(bed_trigger_level).state) {
        return true;
      } else {
        return false;
      }

Bedsensor is a voltage value from Force sensitive resistor which is compared to this trigger level.

bkaufx, 2022-09-12:
I think you should be using template number in this situation, in which case the value will be saved.  Is there some reason that using a template number is a problem for you that I'm not seeing?

bkaufx, 2022-09-12:
You appear to already using a template number.  You just need to add restore_value: true.

closed 2022-09-13 by ssieb