esphome/feature-requests #1848: Support more audio codecs, not just lossy MP3

nagyrobi, 2022-09-01:
**Describe the problem you have/What new integration you would like**
Media Player currently seems to support only MP3 streams. MP3 is a lossy format, sound quality is degraded.
[Home Assistant supports](https://www.youtube.com/watch?v=SEH-DxOsywg) the [Music Assistant](https://github.com/music-assistant/hass-music-assistant) integration which by default renders the audio streams in FLAC, which is a lossless format.
Would be nice to enable support for other codecs too:
- lossless FLAC
- ncompressed PCM up to 48k stereo
- open source OPUS (which is lossy, but way better than MP3)

**Please describe your use case for this integration and alternatives you've tried:**
Since only MP3 is supported, one needs to use other firmware than ESPHome, like squeezelite. The goal would be not to have to rely on third party stuff, but remain in the ESPHome ecosystem.

**Additional context**
Adding more codecs of course needs PSRAM for ESP32, which is not a problem as plenty of such modules are available. This just needs to be pointed out properly in the documentation.

Hedda, 2024-08-06:
FYI, stumbled on this new "***voice-kit***" GitHub repository where ESPHome developers are developing new or improved components for I2S audio (XMOS) support and a new native media player with support for FLAC, WAV/PCM, etc. for the upcoming voice-kit hardware from Nabu Casa:

- https://github.com/esphome/voice-kit/
  - https://github.com/esphome/voice-kit/tree/dev/esphome/components
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/media_player

They already added features and functions or improvements/enhancements to ESPHome, such as:

* *New: Nabu Media Player - new "nabu" media player from Nabu Casa running natively on ESP32*
  * *[Music Assistant](https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/) streams work (both mp3 and flac), but since it requires resampling, the audio quality isn't great*
* *New: Added support for FLAC files*
* *New: Added a proper WAV decoder (that parses WAV headers with LIST, INFO, etc. chunks.)*
* *New: Initial support for playing back local files*
* *New: Playback Control for the VoiceKit*
* *New: Added an `is_paused` condition for media players.*
* *New: Add Click to Converse to button*
* *New: LED animation*
* *New: Scripts for controlling LEDs*
* *New: Update Button Behaviour for the Voice kit*
* *New: Dial Volume Control*
* *New: Timer basic implementation*
* *New: Dial Volume Control*
* *New: Added HTTP(s) OTA updates*
* *New: Dial Volume Control*
* *New: Added Buttons for force ota update.*
* *New: Software Mute Switch*
* *Improvement: A basic resampler adjusts sample rates*
* *Improvement: Configurable output sample rate (for experimental 48kHz XMOS firmware)*
* *Improvement: The DAC mute state is read on boot*
* *Improvement: volume/mute control via the DAC (the wheel works for increasing/decreasing volume)*
* *Improvement: Logs what element failed if the pipeline breaks*
* *Improvement: Fails gracefully if the incoming stream can't be processed*
* *Improvement: Differentiate between user facing LED Ring and Internal LED ring*
* *Point external component to dev branch*

They also have many TODO inline coments in the code there if anyone are interested in helping them:

https://github.com/search?q=repo%3Aesphome%2Fvoice-kit%20todo&type=code

Note! Be aware that there are many comments there to that most of the new stuff are not yet stable.

Hedda, 2024-08-06:
> ethic

ethic? I just wanted to inform about the on-topic open source development in public repository under the ESPHome organization.

https://github.com/esphome/voice-kit/

The news about that upcoming voice-kit hardware platform by Nabu Casa is already on the Home Assistant roadmap so why would it not be developed in the open?

https://www.home-assistant.io/blog/2024/06/12/roadmap-2024h1/#current-priority-2-make-assist-easier-to-start-with

Paulus Schoutsen originally revealed on their [Home Assistant’s ESPHome Summer Release Party on YouTube that Nabu Casa’s ESPHome developers and hardware developers are in fact working on an upcoming open-source voice-kit hardware platform for voice-assistant development that will based on ESP32-S3 in combination with an XMOS xCORE chip

https://www.youtube.com/watch?v=hbcGz3ZlUX4

The upcoming ESPHome-based voice-kit hardware was also mentioned by Mike and Kevin during the Voice Chapter 7 livestream:

https://www.home-assistant.io/blog/2024/06/26/voice-chapter-7/

---

### _Voice assistants_

_Since last year, we have built our voice assistant framework from scratch with our [“Year of the Voice”](https://www.home-assistant.io/blog/2022/12/20/year-of-voice/) initiative. Now that the infrastructure is in place, we want to make sure that it will be usable for everyone (before the demise of Alexa and Google Assistant 😜)._

#### _Current priority 1: Improve Assist capabilities out of the box_
_Our [research has shown](https://community.home-assistant.io/t/poll-what-do-you-use-your-voice-assistant-for-what-do-you-expect-it-to-do-multiple-selections/693669) users are most interested in us improving out-of-the-box capabilities of Assist, for instance, timers, reminders, and music controls._

#### _Current priority 2: Make Assist easier to start with_
_At the moment, there are several things you need to install or configure to get started with voice. We want to make it easier to set up and onboard. There are already some good hardware choices to start using voice, but **we’re exploring building our voice satellite hardware to create a more plug-and-play experience.**_

Hedda, 2024-08-19:
FYI, also check out new "ESPHome media proxy" feature for auto-converting of formats being added upstream to Home Assistant:

* https://github.com/home-assistant/core/pull/123254

As well as these related "MediaPlayerSupportedFormat" patches for ESPHome, ESPHome API, and new voice-kit components that are in development which can tell the "ESPHome media proxy" which MediaPlayer format(s) that ESPHome supports natively:

* https://github.com/esphome/aioesphomeapi/pull/925

plus

* https://github.com/esphome/voice-kit/pull/55

Again, I just want to inform about on-topic open source development in public repository under the ESPHome organization.

Hopefully those will make it to Home Assistant 2024.9 and ESPHome 2024.9 so this new feature could be made use of after that.