esphome/feature-requests #1852: add edge settings on pulse_meter

deathfrozen, 2022-09-04:
**Describe the problem you have/What new integration you would like**
The configuration of the edge for this integration is the same as in pulse_counter.

**Please describe your use case for this integration and alternatives you've tried:**
I am using the pulse_meter component to calculate energy and power consumption in an outdated energy meter. But my counter is constantly shining with an LED, interrupting the glow in the pulse. The pulse_meter component measures along the rising edge, and gives incorrect readings. I need to measure by the falling edge.

**Additional context**

cstaahl, 2023-01-04:
Hi!
You can have the pulse_meter count negative pulses by configuring the pin to invert the signal like so

```
sensor:
  - platform: pulse_meter
    pin:
      number: D5
      inverted: true
      mode:
        input: true
        pullup: true
    internal_filter: 90ms
    internal_filter_mode: PULSE
```

deathfrozen, 2023-02-06:
Thank you, it really helped!

closed 2023-02-07 by deathfrozen