esphome/feature-requests #1853: Support for Waveshare ILI9486

workingmanrob, 2022-09-04:
**Describe the problem you have/What new integration you would like**

Pulling most of this info from https://github.com/esphome/feature-requests/issues/1102 - opened by @EpicLPer - which was closed after adding the xpt2046 but this is a request to support the display - ili9486

In the Dev version of ESPHome support for standard 320x240 ILI9341 based displays has recently been added, see #51. A big thanks to the devs for this already :)
Waveshare has a 3.5" (480x320) and 4" (800×480) version based on the same controller (and/or XPT2046? Not entirely sure where the difference lays here) which sadly does not work with this.

**Please describe your use case for this integration and alternatives you've tried:**

I tried the standard ILI9486 M5stack and TFT_2.4 options already but got no usable output, however the display does seem to read SPI data since, on M5stack, it constantly tries to refresh the display but seems to fail and on TFT_2.4 it very slightly flickers the whole screen. In both cases it never draws anything on screen tho.

**Additional context**

Waveshare offers their Arduino libraries for both TFT versions here: https://github.com/ImpulseAdventure/Waveshare_ILI9486
I also found another library by „GoodTFT“ which also seems to offer one here: https://github.com/goodtft/LCD-show

There was also a PR to support this display opened by MasterTim17 that was not merged to main.  https://github.com/esphome/esphome/pull/1523

nielsnl68, 2022-10-05:
This is now included in my ili9xxx PR at https://github.com/esphome/esphome/pull/3795 check it out and let me know of it is working for you.

workingmanrob, 2022-10-05:
@nielsnl68 not great.. compiles but boot loops with:

[E][display:022]: Could not allocate buffer for display!

minor changes to my working yaml - commented out led_pin and changed the display type - oh and the PR I pulled from obviously

I'm using an esp32 something devkit1..? board: esp32doit-devkit-v1 - that one.

Any suggestions?  I end up in safe mode with the touchscreen not working and a blank display.

workingmanrob, 2022-10-06:
This is what I am using with the PR from MasterTim17..

```
external_components:
  - source: github://pr#1523
    components:
      - display
      - ili9486
```
and the display component

```
display:
  - platform: ili9486
    model: TFT_3.5
    cs_pin: 2
    dc_pin: 15
    led_pin: 21
    reset_pin: 4
    rotation: 90
    update_interval: 1s
    lambda: |-
             do stuff...
```
Not sure if there's another update I need to pull to get this working but I was testing with this:

with and without display
```
external_components:
  - source: github://pr#3795
    components:
      - ili9xxx
      #- display
```
and the change to the display

```
display:
  - platform: ili9xxx
    model: TFT_35
...
    #led_pin: GPIO21
```