esphome/feature-requests #1857: Location Triangulation (ESPresence alternative)

DellanX, 2022-09-09:
**Describe the problem you have/What new integration you would like**

I'm hoping for a method to triangulate my location without needing duplicate ESP32s in the same room.
#1412  points out ESPresence as a solution; however, it doesn't seem like the developers of ESPresence are interested in moving their code logic here. 

For my specific use case, I've been manually creating MQTT messages in lambda expressions; however, that implementation is quite limited in extendibility.

Now that Bluetooth proxy is a thing, I have an alternative idea that might be able to allow for the ESP32s to participate in the triangulation of users.

**Please describe your use case for this integration and alternatives you've tried:**

My idea and request is in three parts. (Two of which would be on the Home Assistant side, and in need of a FR there)
1. Similar to what the MQTT implementation was using. (see below)
During bluetooth proxy events, send the device name or MAC address to home assistant with the RSSI information
2. In Home Assistant, add onto the existing Bluetooth integration support for tracking the MAC <-> RSSI pairs
3. Either in the Home Assistant Bluetooth integration, or some other 3rd party integration
Handle all the Kalman filtering and triangulation

Alternatives I've tried:
1. MQTT Room, this didn't work at all for me, as my floorplan is very irregular in shapes, making it difficult to determine what room I am in by which device I am closest to.  [#3759](https://github.com/esphome/esphome/pull/3759) would not be helpful for me
2. ESPResence, it works, but I cannot used sensors such as capacitive touch or RFID readers. (#1412 touches on this a bit)
3. MQTT lamda script, also works, but difficult to maintain. Requires MQTT broker
4. There was a github project I found by [rpatel3001](https://github.com/rpatel3001/BleDistance) attempting to recreate espresence; however, it required a fixed list of UUIDs to be compiled, and thus wasn't very dynamic 

**Additional context**

I think the easiest way to do this would be to add context information to the Bluetooth Proxy, which can be handled on the api side of homeassistant. Simply, the MAC of the ESP32 and the RSSI being passed upstream with the BLE proxy calls.
(That or the device name of the esp32)

This way, Home Assistant can handle all of the filtering and triangulation (Likely an integration that would need to be developed on Home Assistant's side)
Having filtering on the ESP32s would be nice; however, I'd argue that the memory of the ESP32 would be better spent elsewhere.

nagyrobi, 2022-09-09:
I think part 1 of your idea is already in. If you've set up multiple ESPHome nodes as proxies, take a look of you enable debug in HA:
```
logger:
  default: warn
  logs:
    homeassistant.components.bluetooth: debug
```
You'll see that data is pouring in: proxy which passed it through, source of the data (incl MAC) and RSSI value are logged among the data itself.

So this request is actually more appropriate to be raised in Home Assistant...

DellanX, 2022-09-09:
I played around with the logs, but only see the MAC of the bluetooth device, not the MAC of the Bluetooth Proxy(ESPHome).
I'm think I may have misspoke. To clarify, the request above is to add ESPHome's MAC addresses (or ESPHome device name) into the proxied message.

Below is one of the responses detected by HA, it has the MAC address of the found device, but I don't see the mac address of the proxy device. 

```log
2022-09-09 12:39:14.456 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0: F1:E1:30:33:37:B8 AdvertisementData(service_data={'0000feed-0000-1000-8000-00805f9b34fb': b'\x02\x00\xea-q\x17\xa5D\x82\x8a'}, service_uuids=['0000feed-0000-1000-8000-00805f9b34fb']) connectable: True match: set() rssi: -68
```

I will agree, that the largest portion of this task will be on the HA side, and will need an issue there; however, the proxy device identification information will be required in some form to do anything there. This Issue is for the ESPHome to provide such data.

nagyrobi, 2022-09-09:
It's `hci0` that denotes the device which received the package. In your case this is likely a local adapter. If you had used an ESPHome Bluetooth Proxy, you'd see the ESPHome node name there. If multiple, then each.

DellanX, 2022-09-09:
Oh wow! I see it now (Definitely just assumed that was the HA interface receiving the BT packet, hci0)

Looks like the ESP device name is populated there, and that this portion of the Feature Request has already been handled!
```log
2022-09-09 13:43:20.181 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] tracer-kitchen: 6A:37:F8:E0:5F:9F AdvertisementData(manufacturer_data={6: b'\x01\t \x02\x13"&E\xf7\x8b\xc8c\x00r\xf1t\x18\xc4\x88\xa8\xf2I\xfeo$\x0c\xa0'}) connectable: False match: set() rssi: -80
```

I'll close this issue, and open up one on the HA side next. Thanks!

closed 2022-09-09 by DellanX

nagyrobi, 2022-09-09:
Please post here the link to the FR you make in HA for this, for easier tracking.

DellanX, 2022-09-09:
The HA FR forum thread I create is here:
https://community.home-assistant.io/t/bluetooth-position-triangulation-gps-ips-espresence/459167