esphome/feature-requests #1859: Add humidifier component

Brokly, 2022-09-09:
**Describe the problem you have/What new integration you would like**

I did the humidifier control on ESP32. It turned out that the humidifier is in the HOME ASSISTANT, but is missing in ESPHOME. At the moment I use my component as a set of controls, without a single card.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to use standard methods of controlling and displaying the humidifier in the HOME ASSISTANT interface

**Additional context**

nagyrobi, 2022-09-09:
See: https://github.com/esphome/esphome/pull/3523

paravoid, 2024-05-02:
See https://github.com/esphome/esphome/pull/6591 for a newer attempt.

paravoid, 2024-05-09:
> See [esphome/esphome#6591](https://github.com/esphome/esphome/pull/6591) for a newer attempt.

Superseded by esphome/esphome#6678.