esphome/feature-requests #1861: Enhance CAN BUS integration for low speed

MaKo-RS, 2022-09-10:
**Describe the problem you have/What new integration you would like**

I would like to have an enhancement to the existing CAN integration for **esp32_can** to be able to run lower baudrates like 20kbps.

**Please describe your use case for this integration and alternatives you've tried:**

In industrial or building automation systems which run CANopen protocol or similar, the default baudrate is 20kbps. Due to long bus lines it is often not possible to run higher baudrates. 
One possible solution is to use the external CAN device MCP2515 via SPI. But the internal CAN engine in ESP32 is able to do the same. To reduce hardware the idea is to use the internal engine for this.
 
In ESP32 devices with version lower V2 it is not possible to use low speed. The APB clock runs with 80MHz and the prescaler for the bit timing is only 6 bits wide. From Version V2 of ESP32 one can enable an additional divider by 2 and low speeds are possible. Since 2 or 3 years V3 from ESP32 is available. Enabling the divider is done with setting CONFIG_ESP32_REV_MIN to 2 or higher.

In YAML configuration of ESP32_CAN I would prefer to have a possibility to select baudrates as it is today. Additional it would be nice to be able to overwrite the defaults for the bittiming registers like that:

>canbus:
>&nbsp;&nbsp;- platform: esp32_can
>&nbsp;&nbsp;&nbsp;&nbsp;tx_pin: GPIO5
>&nbsp;&nbsp;&nbsp;&nbsp;rx_pin: GPIO4
>&nbsp;&nbsp;&nbsp;&nbsp;can_id: 160
>&nbsp;&nbsp;&nbsp;&nbsp;bit_rate: 20KBPS
>&nbsp;&nbsp;&nbsp;&nbsp;...

Or more felxible for special bus systems:

>canbus:
>&nbsp;&nbsp;\- platform: esp32_can
>&nbsp;&nbsp;&nbsp;&nbsp;tx_pin: GPIO5
>&nbsp;&nbsp;&nbsp;&nbsp;rx_pin: GPIO4
>&nbsp;&nbsp;&nbsp;&nbsp;can_id: 160
>&nbsp;&nbsp;&nbsp;&nbsp;divider2: true
>&nbsp;&nbsp;&nbsp;&nbsp;btr0: xxx
>&nbsp;&nbsp;&nbsp;&nbsp;btr1: xxx
>&nbsp;&nbsp;&nbsp;&nbsp;...

**Additional context**

CAN description @espressif:
[https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/can.html](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/can.html)
In section **Bit Timing** is the desription.

bakermat, 2023-02-03:
I'm also looking for this, as a few of the devices I'm working with have a 20kbps bitrate. Would be great if this can be added to esp32_can.

kala79, 2023-07-10:
This feature would also be desirable for me!