esphome/feature-requests #1862: Bluetooth only device with BTHome format

stumpylog, 2022-09-11:
**Describe the problem you have/What new integration you would like**

It would be awesome to be able to configure an ESP device with only Bluetooth, which sends sensor data using the [BTHome](https://bthome.io/) data format for relay to Home Assistant using the new proxies.

This would be very beneficial for battery powered devices, while still leveraging the YAML configuration and existing components (probably just sensors).

**Please describe your use case for this integration and alternatives you've tried:**

The only current alternative is writing a BTHome and BLE setup with the ESP-IDF or Arduino libraries manually.

**Additional context**

This may not fit into the normal ESPHome setup, but be better as another project, which shares as much as possible with ESPHome's already existing code.

stumpylog, 2022-10-02:
In case anyone would like to roll their own so to speak, I have a working version: https://github.com/stumpylog/bthome-weather-station

It uses a BME280 and an ESP32 to send values to Home Assistant.  It's nowhere near the user-friendly ESPHome, but it also isn't so hard to use.

digaus, 2023-02-17:
Wonder why this is not a thing yet. 

Combined with the XIAO ESP32C3 we could then make tiny door/window sensors with very low power consumption which can even fit inside the window frame.

Really interested in this feature☝🏻

jnimmo, 2023-10-16:
Really interested to see this too.
One step further, we could consider Home Assistant App being able to proxy BTHome advertisements through to HA also (my use case is a CANbus board to broadcast the current car charge status).
Some interesting relevant work here - just missing the BLE transmitter
- https://github.com/afarago/esphome_component_bthome

voed, 2024-08-09:
Would be really cool feature