esphome/feature-requests #1863: Ginlong Solis solar inverter: RS485 'COM' connection

grob6000, 2022-09-12:
**Describe the problem you have/What new integration you would like**

I would like to add a component to connect an esphome device directly to my Ginlong Solis solar inverter, using the RS485 COM port on the bottom (used by their proprietary wifi stick), to collect and send data to home assistant directly within my local network (eliminating current dependence on the solis cloud service).

I have developed an external component that works for my specific setup, as a piggyback on the existing wifi stick. I've worked out one specific inverter's protocol. However, I'd be interested in tidying this up and building it into esphome if others had interest in a similar application, and may need to plan for this to support multiple protocols from the outset (I'm pretty sure there's quite a variety among solis inverters).

https://github.com/grob6000/esphome-externalcomponents

What I propose is to build the above into esphome as a built-in component, with:
- A config line for 'protocol' in case there are more out there that I haven't decoded yet - and these can be implemented over time.
- Also, my current implementation requires the manufacturer wifi stick to be present to work - I'd need to extend this to allow periodic sending the request messages (with a config line to select master or piggyback mode).

I would also appreciate feedback from the community (and others who might be looking for something similar) about the above and how you'd like to see it implemented.

**Please describe your use case for this integration and alternatives you've tried:**

I've tried the home assistant soliscloud integration (via HACS). This works well! However it has a couple of fundamental drawbacks for me:
1. Use of an external cloud service (dependence on this service, internet connectivity to function)
2. 5-minute or so frequency of datapoints
3. 1~5 minute delay of data between the actual state of the inverter and the sensor state in home assistant (due to the long path travelled: inverter --> rs485 --> solis wifi stick --> solis cloud --> soliscloud api --> home assistant).

**Additional context**

My supposedly smart mains meter has a blinking light which blinks on both import and export (no way to know which though!). In my use case, the concept was to use the solar inverter's generation data and consumption data from another current transformer meter as a way to calculate the momentary import/export power from my house. I can use the relative size of the solar and consumption power values to determine the import/export direction at any time, and count the pulses of the mains meter to get a figure for import/export kWh pretty close to what my power company sends me in the post once a month: but LIVE! However, critical in this case is that the solar generation and consumption measurements are reliable and highly contemporaneous. With rapidly changing solar generation (e.g. clouds), a few minutes of incorrect direction would count a fair number of pulses in the wrong direction and lead to error.

grob6000, 2022-09-12:
To see what you might gain from this, see here the difference between the HACS soliscloud integration data (orange) and the data from my serial esphome device (light blue). This is a cloudy afternoon. Note the rise/fall of the inverter power is significant in the order of minutes, and the healthy smoothing/lag seen in the data from the soliscloud.

I'd suggest the soliscloud data makes nicer graphs for people with a passing interest, but using the power data for any kind of near-real-time application would be ill-advised:

[image]

nagyrobi, 2022-09-12:
Please contact with the dev channel on Discord to ask for directions on how to prepare this as a PR to ESPHome.