esphome/feature-requests #1864: Port i2s_audio to idf framework

JeeCee1, 2022-09-12:
**Describe the problem you have/What new integration you would like**

It would reduce flash size if the `i2s_audio` integration could be ported to use the `idf` framework.

**Please describe your use case for this integration and alternatives you've tried:**

Combining `i2s_audio` with `bluetooth_proxy` fills up the flash, resulting in crashes.
See [this](https://github.com/esphome/issues/issues/3481) issue.

**Additional context**

martijnve, 2022-09-17:
"Combining i2s_audio with bluetooth_proxy fills up the flash, resulting in crashes."

I don't know anything about the proposed solution but came here to look for this problem. Combining a small media player and a ble hub would be very nice.

elik745i, 2023-07-23:
face the same issue, please help to resolve

ChristophCaina, 2023-10-14:
here, too :)

alfwro13, 2023-10-19:
face the same issue please progress this

formatBCE, 2024-05-23:
I'm using this with my S3: 
https://github.com/gnumpi/esphome_audio
It works good!

closed 2025-02-10 by jesserockz