esphome/feature-requests #1865: Support for DS2438      1-Wire Battery Management Smart Battery Monitor with Temp Sensor

zillawilli, 2022-09-12:
**Describe the problem you have/What new integration you would like**

I request ESPHOME team to implement support for DS2438 1-Wire Battery Management Smart Battery Monitor with Temp Sensor.

**Please describe your use case for this integration and alternatives you've tried:**

The DS2438 is a Smart Battery Monitor and is used in its original function as a mobile phone battery monitor.
For automation, especially in the **_smart home area_**, it is also used for other functions:

    as voltage meter (supplies the bus voltage (VDD)
    as thermometer - the temperature is measured
    AD-Converter (VAD) - is used to measure humidity or brightness
    Current input (XSENS) - freely available

**Additional context**

Type/designation | DS2438Z+
Housing | SOIC-8 (SMD)
Supply voltage | 2.4 V to 10 V
Function | Temperature sensor
Analog Input 0-10VDC (VAD)
Analog Input 0-250mV,
+/- (Xsens) Measurement Operating voltage (VDD)
Operating temperature | -40°C to +85°C
Details | See data sheet 
[DS2438.pdf](https://github.com/esphome/feature-requests/files/9549921/DS2438.pdf)
[DS2438 current.pdf](https://github.com/esphome/feature-requests/files/9550022/DS2438.current.pdf)
[image]

Thank you very much for your great work!!!!!

luftdieb, 2024-05-26:
I would like to announce also my interest in this feature. Would be great, if it would be easily supported :-)

randybb, 2024-05-27:
This is great as many analog sensors (temp, hum, light,...) can be connected easily to 1w bus. There is already a product https://sedtronic.eu/en/unica-1wire/30-52-unica-1-wire-tvsl.html heavily used in Loxone home automation systems.

ukuehn, 2026-09-06:
In case this request is still relevant please be informed that I created an external component to support the ds2438. You'll find it at https://github.com/ukuehn/esphome_external_components. I wrote it to support a humidity sensor very similar to the schematic above (with a HIH-5030 instead of the HIH-4030). 

In case this proves useful I'd appreciate any feedback you might be able to provide, in particular if you encounter any issues.