esphome/feature-requests #1867: way for sml / obis to accept  1.8.0 instead of 1-0:1.8.0 (so no prefix)

evlo, 2022-09-13:
**Describe the problem you have/What new integration you would like**

I'm trying to migrate from tasmota, i just like esphome better, but the smart meter interface is still impossible to me as my meter uses obis codes as espohome recognises. Ie. "1-0:1.8.0" is what esphome except as obis, but "1.8.0" is what I get from electrical meter. Do you know how this could be solved with some workaround?

output from tuya
[image]

**Please describe your use case for this integration and alternatives you've tried:**

I tried various custom OBIS reader external components

**Additional context**

Also to wake up meter the string /?! (hex 2F3F210D0A) at 300 baud, but it seems that rest of the communication could be done on 9600

nagyrobi, 2022-09-14:
Please ask on Discord for faster answers.

TheJulianJES, 2022-11-17:
@evlo I'm basically having the same issue as you. Did you manage to solve/work around it?

FYI, also asked on Discord: https://discord.com/channels/429907082951524364/429907082955718657/1042909407564013638

evlo, 2022-11-17:
https://github.com/evlo/esphome_ZPA-ZE312

*https://github.com/jplitza/esphome_components*

TheJulianJES, 2022-11-17:
@evlo Yep, I actually found your component in the mean-time! My meter sends a lot of other data that cause "errors", but total consumption + current consumption is working fine now!
Thanks for your work!