esphome/feature-requests #1871: General purpose PID-Controller (or something similar)

RubenKelevra, 2022-09-16:
**Please describe your use case for this integration and alternatives you've tried:**

I'm using ESPHome to control the fan speed of an [Ikea Förnuftig air cleaner](https://www.ikea.com/ch/de/p/foernuftig-luftreiniger-weiss-50461937/) and read the sensor data of an [Ikea Vindriktning PM2.5 sensor](https://www.ikea.com/ch/de/p/vindriktning-luftqualitaetssensor-70498242/).

To control the air cleaner with the sensor data I'm using a PID-Controller (a custom component) inside of Home Assistant.

This solution isn't stable.

**Describe the problem you have/What new integration you would like**
I would really like to access just the sensor data via Home Assistant as Hub with the air cleaner and have a general purpose PID-Controller which runs on the ESP8266.

The [Climate PID Controller](https://esphome.io/components/climate/pid.html) could maybe be repurposed for this, but I'm really not sure if reading PM2.5 particle counts and reinterpret them as °C is the right thing to move forward here.

**Additional context**

nagyrobi, 2022-09-16:
> To control the air cleaner with the sensor data I'm using a PID-Controller (a custom component) inside of Home Assistant.
> This solution isn't stable.

In what aspect is not stable?

davey, 2022-12-01:
I'd like to confirm that need for a general purpose / generic PID controller: currently the Climate PID controller can only work with a temperature/°C, but it would be really great to be able to use it for e.g. basic humidity control - or basically for anything with a float sensor and one (e.g. increase something, like heating) or two (increase and decrease something, like heating and cooling) float outputs.

SeByDocKy, 2022-12-09:
Definitively a must to be done for ESPhome !!!!

nagyrobi, 2022-12-09:
A quite complex thing to think about. The current PID climate uses the _ziegler-nichols_ tuning rule, which seems good for temperature regulation.

This may not be appropriate for other applications, several other rules exist like _tyreus-luyben_, _ciancone-marlin_, _pessen_ etc. A generic PID controller should be able to be parametrised in a very flexible way, to support any kind of rule, even completely custom.

<br class="Apple-interchange-newline">

RubenKelevra, 2022-12-14:
> > To control the air cleaner with the sensor data I'm using a PID-Controller (a custom component) inside of Home Assistant.
> > This solution isn't stable.
> 
> In what aspect is not stable?
> 

I would consider it an alpha/beta not stable.

patrickcollins12, 2022-12-28:
I'm happy to look into it. Could be challenging because PID is a subclass of Climate and so assumes much more than just temperature (cool, warm, etc).

patrickcollins12, 2022-12-30:
> Definitively a must to be done for ESPhome !!!! Planning to use ESPhome for solar power diverter regulation. Here my target will be current (A)

@SeByDocKy what would you want it to do? What would be the sensor and what would it control?

SeByDocKy, 2022-12-30:
> > Definitively a must to be done for ESPhome !!!! Planning to use ESPhome for solar power diverter regulation. Here my target will be current (A)
> 
> @SeByDocKy what would you want it to do? What would be the sensor and what would it control?

the sensor would be either a DC/AC current sensor I controlling the power P output of a AC dimmer.  Mathematiccally speaking,  I would like to do something like 

Max P
st  I_lower_bound<I<I_upper_bound    where I is the DC current input/output of a battery for example.

Let's clarify more. If the system produces the  AC power P_k  (to a pure resistive load) such as P_k < Psolar_k  then I_k > I_upper_bound.... so I can increase P such P_{k+1} = a_P'{k}, a > 1.. If P_{k+1} < Psolar_k, I can continue to increase P else (I__{k+1} <  I_lower_bound), I have to apply P_{k+1} = b_P'{k}, b < 1 and so on ...  

We did that with a pure arduino code : https://github.com/bidouilleurs/routeur_solaire

But I would like to use ESPhome instead...

patrickcollins12, 2022-12-30:
I think we would need a totally new component. GenericPID(?). It would be very similar to PIDClimate in functionality. I need to duplicate to ensure that PIDClimate maintains backward compatibility.

SeByDocKy, 2022-12-30:
> I think we would need a totally new component. GenericPID(?). It would be very similar to PIDClimate in functionality. I need to duplicate to ensure that PIDClimate maintains backward compatibility.

Sound an excellent idea....

patrickcollins12, 2023-02-07:
I've been working on this for a couple nights. I'm probably 30% complete. This is how the config would look. Before I continue, I welcome your input. 

```
# this is the existing climate pid
climate:
  - platform: pid
    id: console_thermostat_old
    name: "Console Fan Thermostat"

    sensor: console_fan_temperature

    default_target_temperature: 21
    cool_output: console_fan_speed

    control_parameters:
      kp: 0.0
      ki: 0.0
      kd: 0.0

# this is the new generic pid, note increase/decrease output

pid_control:
  - id: humidity_controller
    name: "Bathroom Humidity Controller"

    sensor: bathroom_humidity_sensor

    default_target: 60
    decrease_output: bathroom_fan
    increase_output: shower_heat

    control_parameters:
      kp: 0.0
      ki: 0.0
      kd: 0.0

  - id: second_controller
    default_target: 22
    increase_output: increase_auger_speed
    sensor: bbq_sensor
    control_parameters:
      kp: 0.0
      ki: 0.0
      kd: 0.0
```

SeByDocKy, 2023-02-07:
> I've been working on this for a couple nights. I'm probably 30% complete. This is how the config would look. Before I continue, I welcome your input.
>
> ```
> # this is the existing climate pid [...]

Sound great and clear for me :)

SeByDocKy, 2023-02-08:
Thinking about your futur general PID controller, came in my mind three additions (that are important for my futur project with):
i) the possibility to enable/disable a PID_controller , for example with  (public) methods $id_pid.set_enable/$id_pid.set_disable or with a internal state that can be changed with lambda. I have to explain: I plain to code a general solar power diverter for both ongrid ior offgrid. The first one need to control the AC current flow while the second need to control the DC current flow. If I define two general PID controllers only one can be active so I need to enable one and disable the second one (and vise and versa)

ii) The ossibility to control (kp,ki,kd) with for example a esphome number object (in slider mode). It would allow to fine tune these parameters  without requiering each time to recompile and flash OTA after a change

iii) and the last  add a (set of) condition(s)  that need to be true to increase the output field  (pointing to some binary_sensors) defined for example like:
pid_condition_to_increase:
  - binary_sensor_id: id_binary_sensor1
  - binary_sensor_id: id_binary_sensor2
 ect ...

.In the same way, condition to recpect to decrease : 

pid_condition_to_decrease:
  - binary_sensor_id: id_binary_sensor3
  - binary_sensor_id: id_binary_sensor4

Why it can be great to have that ?  For my power diverter for grid-tied system, the regulation must be done on the DC current.... but you have to stop to increase the output as soon as also you start to pick up some energy from the grid (inverter if they don't have enough power from solar then from battery will take power then from the grid)

I hope I was enough clear..... Anyway thanks to work on this very important futur new integration

patrickcollins12, 2023-02-09:
Thanks @SeByDocKy I appreciate this input. 

1. I hadn't considered an enable/disable mechanism. I agree this is critical. You get this for free in PIDClimate because all climate devices have an on/off mechanism, so it makes sense to build it in here. I'll add this to the list.

2. I built an example [here](https://github.com/patrickcollins12/esphome-fan-controller/blob/master/console-fan.yaml#L38) of how to wire a Number template to kp,ki,kd. So I think you'll get this functionality for free. Although it is 1-directional. Thoughts?

3. This feels specific to your use case and not common in PID controllers. I would encourage you to use the enable/disable feature to control this. Thoughts?

SeByDocKy, 2023-04-04:
> Thanks @SeByDocKy I appreciate this input.
>
> 1. I hadn't considered an enable/disable mechanism. I agree this is critical. You get this for free in PIDClimate because all climate devices have an [...]

Hello, any news of ypur general PID developpent ?

Sébastien

SeByDocKy, 2023-04-23:
> I have not made further progress since I started a new job. But I hope to get back to it soon.

Great news :)

Tuckie, 2024-06-13:
Any updates on this? Looking for something to regulate water pressure with a pressure sensor and a pump.

bzumik1, 2024-06-22:
Hi, also loooking for something similar :)

shufps, 2024-12-03:
> currently the Climate PID controller can only work with a temperature

Any update on this? I need humidity :pleading_face: :pray:

bzumik1, 2024-12-03:
Also lookin for this, I need to regulate CO2 🌬️ concentration.

adamja, 2024-12-19:
+1 for this feature. I feel like this is an important missing piece to the ESPHome component ecosystem! If I had the ability I would have a crack at it... 😄 ESPHome
@patrickcollins12 we need our champion back!! Hope the job is going well though

RubenKelevra, 2024-12-19:
@patrickcollins12 do you got time to complete this, or should I take over? 🤔

patrickcollins12, 2024-12-20:
I originally tried to make a base class for PID from which the PIDClimate and the new PIDController would inherit. It got too hard... andI failed/quit. 

The easier way is to copy PIDClimate and create a new class called PID. Might be a lot easier to make, but has some replicated logic in the ESPHome code base.

adivinus, 2025-03-01:
+ 1 for that. 
I need it for several purposes:
- regulate CO2 level in the rooms;
- regulate temperature; 
- regualte air humidiry;
- regulate irrigation for plants; 
- regulate gas in grill oven;
. . . 

Atd all of them could be reached by very easy abstract component.

RubenKelevra, 2025-03-01:
Hey @patrickcollins12 would you mind if I take over? :)

samvel-galstyan, 2025-03-01:
+1, needed to control pump of water, depending on pressure sensor.

patrickcollins12, 2025-03-02:
@RubenKelevra  please go ahead! Let me know if I can help.

RubenKelevra, 2025-03-02:
Thanks, hope my changes will be merged this time 🥲

patrickcollins12, 2025-03-02:
You have to persistently remind. I had to ask kindly about 10 times until it got merged... the core team don't not much about pid so get a little worried merging changes to it.

RubenKelevra, 2025-03-02:
I already have to f[ight bots for 3 of my PRs](https://github.com/esphome/esphome-docs/pull/4351#issuecomment-2664639265), so the doc PRs don't get closed automatically... 🙄

patrickcollins12, 2025-03-02:
I had to chase them in discord in the developers channel.

RubenKelevra, 2025-03-05:
Given the wide variety of applications people quoted here, I think PID isn't really suitable for those tasks, tbh.

To give a short overview for which applications PID is useful: If you want to regulate a single entity, based on one parameter, while the characteristic is non-linear but not changing over time.

This may have very limited applications, but even a simple climate PID is actually not really seeing the full picture.

I think the better solution is to switch to adding MPC to ESPHome, which will use significantly more memory and processing power but provides multi-input, multi-output, and multi-target control. MPC continuously optimizes control actions based on an internal system model and previous control steps. Unlike PID, it predicts future system behavior and makes preemptive adjustments instead of reacting to errors.

#### Example #1:
Say you provide it with:
- Solar output forecasts
- Live solar output
- Power meter readings (grid & self-consumption)
- presence detection
- Time of day & day of the week
- Battery state, battery charging rate
- Individual power meter readings for high-consumption devices

By optimizing the charging current, MPC can minimize grid power consumption and maximize self-consumption efficiency. Unlike PID, it anticipates demand changes and adjusts preemptively.

#### Example #2:
Inputs:
- Outdoor temperature, humidity, air pressure, sun radiation
- Room temperatures (via dedicated sensors)
- Valve positions of all radiators
- Air Pressure Outside
- Wind / Precipitation / Outdoor Temperature forcast
- Local temperature of radiator controllers
- Window & door statuses
- Sunshade position (open/closed)

Outputs:
- Control over the radiator set points

MPC will automatically determine the optimal heating strategy based on past system behavior, external conditions, and future predictions. It will even recognize which sensor values are useful or redundant over time.

#### Drawbacks compared to PID

- MPC requires solving an optimization problem at every control step (high processing demand)
- On ESP32-S3, simplified models and shorter prediction horizons are necessary.
- Output Constraints are required to avoid excessive switching (e.g., relay wear).
- There may be very unstable control behavior if there's no processed input data, which slowly improves over time
- We need a way to reset the control behavior, so it will "relearn"
- There need to be a way to force certain outputs when necessary (like turning the heat off if a window is open)

@patrickcollins12 what do you think?

RubenKelevra, 2025-03-05:
So I've invested some time into implementing a rough version of MPC, while still at early tinkering stage I wanted to write a short update:

So after looking at much data in HA and how it may affect systems I like to use this for, I determined that the main limiting factor will be the small memory.

MPC typically uses recent input data to learn and predict how systems respond to different actions. 
While this may work fine for some applications, I see quite a gap here in adapting abilities. In real-world home automation, some system reactions are delayed—like when sunlight enters a room, first affecting the light sensor readings and only later increasing the room temperature.

If such events are very sparse and delayed, MPC may not be able to determine the connection between both, and thus won't predict that the room heats up.

To close the gap it would make sense to store more data, a lot more data, but that's not possible due to the restricted memory.

So the solution I came up with is to evaluate prediction accuracy continuously. When MPC makes a prediction that doesn't match actual sensor readings (for example, not anticipating the room temperature increase despite increased brightness), it indicates our prediction model missed something.

Instead of discarding this discrepancy, I use the new input, instead of the old prediction, and reverse the prediction process, to find which input data may be responsible for our bad prediction. Ideally, we would see a rising Lux value here as fitting the old prediction, but not the new sensor value.

By assigning greater weight to this signal, I then rerun the prediction forward, to see if the error has improved. If that's the case, this signal is stored in the PSRAM.

I've started to run some simulations with that to see how successful this whole approach is, as it's quite costly. So there's a balance between doing an analysis to determine the root cause for a prediction error.

To lighten the load it would make sense to also store "false" signals which will be determined as potential candidates, but did not improve the prediction in the past.

The whole memory is currently maybe not ideally structured as sorted list with a binary search and simple replacement of the entries with the worst improvement of error if it reaches the maximum length, plus some simple decay. The logic for decay will just remove entries not read or updated for 256 days in bulk, to avoid storing elements forever which were just single events.

There's also no permanent storage or load into PSRAM yet, so on each reboot the "long term storage" would be reset, which is not ideal.

I'll try some simulations in the next days and maybe throw it onto an ESP32, if the performance looks promising on a desktop.

patrickcollins12, 2025-03-06:
Very interesting. I think this is worth pursuing as well as, and separate to, the generic PID system.

RubenKelevra, 2025-03-06:
@patrickcollins12 I don't think generic PID is really necessary, as it can also handle single input single output, and the more complex long term history could be turned off.

I did some very rough calculations and without specific ESP32 optimizations it should be able to complete a cycle in 42 ms.

With S3's vector processor it should be much faster, in the region of 2 ms.

So if I haven't overlooked something here, it should work fine even on low end hardware, as long as there's a FPU on board.

ESP8266 might be a bit overwhelmed by the amount of necessary floating point operations.

SeByDocKy, 2025-03-06:
IMHO, while MPC looks promising to do better "job" with a deeper overview of the scene, a generic PID is already a good start.... Don't forget also the PID won't run standalone in the ESP... For example, I got a project running 628 entities on a S3.... with tons of Uarts, I2C, SPI exchanges... 
FIY, I already wrote a first version of a PID dedicaced for solar routing..... Working great already but I want to rewrite in a more natve way (in terme of sensors, select, numbers, binary sensors, outputs)

RubenKelevra, 2025-03-06:
@SeByDocKy well, I've used PID before in HA and regardless how I turned it, it kinda never worked as expected. Now while researching for MPC found what the issue is: PID is not well working for slow reacting systems and only reactive, not proactive.

So even while tuned for a slow reacting system, like controlling an air cleaner based on a sensor it can't handle the slow reaction time of its control inputs, and will basically switch the air cleaner instead of in multiple steps up and down, from the highest step down to off, and then, once the air is dirty again up to the highest setting basically.

I see a very limited use case for a regulation function which needs a lot of manual tinkering or a manual startup for some calibration, and then a recalibration each time the system's dynamic has changed.

MPC on the other hand does adapt to changing system dynamics automatically, and there's no tuning necessary.

If you run a a project with 600 entities, still really don't see an issue running MPC, if it's an S3. I mean, you will loose 2 ms on one core. That's much less time than pushing some states to Home Assistant over wifi. 🤔

patrickcollins12, 2025-03-07:
Let's build both!! MPC seems amazing

RubenKelevra, 2025-03-07:
@patrickcollins12 yeah, in the end this may be a good idea. While I see the benefits of using MPC for everything, having PID as alternative for very high frequent or very lightweight control logic may make sense for some applications.

That said, I wonder if it would make sense to instead build a second and third module, it may make sense to build a single module which contains all the boiler plating, and where you can select the algorithm as parameter.

This would on one hand make sense, as most control algorithms are designed around receiving and outputting floating point numbers, so output limitations like boolean, range, steps etc. and input conversions like boolean to float etc. could be all enforced in this code and after the value returns from the algorithm. 

On the other hand simplifies the code base, as we would only need to add each algorithm itself, instead of rewriting specifically tailored boiler plate code for input and output. Plus some definitions, amount of inputs, outputs and setpoints which are available.

@SeByDocKy would this make sense to you?

the-nicolas, 2025-04-13:
Very nice ideas here, I also use HA and earphone to control my solar system: Charging batteries, heating water, inverting from battery etc.) and some general flexible building blocks would really help here.

Important: The output has to be a float and not just a boolean switch.

Right now I implemented most of the stuff in node-red, but would be great to have some lower level handling these things more bullet proof.

lutorm, 2025-05-10:
I'm also looking for a general purpose PID controller. I was surprised to find that there's a special-purpose thing for temperature but no generic one. It seems backwards to go about it that way. Anyway, I'd be interested if anyone's code it up, otherwise it's not super complicated to implement directly but I'd prefer not.

hsteinhaus, 2025-05-10:
I have built a GP PID controller based on the climate-pid for my project and I am happy to share it: https://github.com/hsteinhaus/esphome-gppid .

The quality is far away from what would be worth a PR, doc is non-existent yet. May be one day...

SeByDocKy, 2025-05-15:
For offgrid solar application, I wrote also a component :  
Here the regulation loop via PID is done on battery current. I plan to write another one for "classic" grid tied solar router

vontrapp, 2025-07-25:
> Given the wide variety of applications people quoted here, I think PID isn't really suitable for those tasks, tbh.
>
> To give a short overview for which applications PID is useful: If you want to [...]

MPC could be a really cool addition no doubt. But it could also be unusable to a lot of people either from hardware restrictions/cost or knowledge or simply being overkill for their application.

MPC should not come as a replacement or "instead of" PID control. PID control is stupid simple and also very powerful. PID controllers can easily respond fast enough to make the difference between preemptive action and reaction insignificant for a lot of use cases.

Implement the generic PID first, then MPC if there's enough interest, as that's going to be a hard lift to make it fit in the ESP hardware constraints as well as the complexity of the component parameters.

adamja, 2025-07-25:
I agree @vontrapp. There should be a base level Generic PID that is one input and one output. I've been working on one that uses this library for the pid calculation logic: https://github.com/br3ttb/Arduino-PID-Library

But simply it will take an input that will need to be scaled so that it's range is 0.0-1.0 (0% - 100%), run the PID logic and output a response in a scaled output between 0.0-1.0 (0% - 100%) with configuration options for P, I, D set_point, etc... This is completely agnostic to what application the PID controller is used for, which I think should be the intension of a Generic PID controller component?

Does anyone have any opinions of the actual PID calculation side of things, I don't really know the depths of the maths behind them, but I did like the forum from around 2011 here with people throwing their two sense in (which is behind the library I linked above) 😄 
http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/

RubenKelevra, 2025-07-25:
@adamja note that a Arduino version is not a good idea, as you would otherwise be restricted to this framework, which is slow and got a lot of overhead, as it's a layer on top of ESP-IDF.

adamja, 2025-07-25:
@RubenKelevra oh good point, it's literally just simple maths in the library, so there is no real dependency on the arduino framework i believe :)

RubenKelevra, 2025-07-25:
If @patrickcollins12 or anyone else here in this thread has some incomplete work: I'm happy to incorporate that. Just send it over via email (address in profile) and I add you as co-author. Just make sure to mention a fitting license. 

I'll try replace the current PID controller in it's feature as well, so we don't end up with two different PID controllers for different purposes, and instead can switch the settings on them.

adamja, 2025-07-25:
This is my half baked project for creating a PID controller. I did trial it with a real application and it was working well! But it's far from polished or wrapped up properly, and has arduino dependencies as you mentioned earlier!

https://github.com/adamja/esphome-pid

It has some cool features, I was thinking of adding a master on/off switch and an input high/low alarm with a callback trigger also. I've never created a "proper" esphome component that I would expect to be added into the main branch - so it's not that rn😄

But the concept is there - that part I am happy with

SeByDocKy, 2025-07-25:
> This is my half baked project for creating a PID controller. I did trial it with a real application and it was working well! But it's far from polished or wrapped up properly, and has arduino [...]

Look pretty good !!!!

RubenKelevra, 2025-07-25:
Okay, short status update: I did my research on different tunings, thought long about different use cases and what kind of output filters, output ranges, output stepping would be necessary for them. And I  drafted a full documentation as guideline for the implementation for me.

So as planned there will be (in no particular order):

- quantization of the output to:
  - steps, like [0, 1, 2, 3, 4] or [0, 33, 66, 99, 100] etc.
  - int with min/max
  - float with min/max
  - float with min/max and step, like 5-35 (°C) as range and 0.5 °C stepping.
  - boolean
- In addition there will be support for "dead bands" of the output. This is important for e.g. controlling PWM. Say you want 0-1 as float with 0.05 steps, but between 0-0.1 your motor stalls, you can configure a dead band for that.
- the module will allow selecting `PID` as algorithm right now, the idea is to expand it with MPC in the future, but use the whole input and output boiler plating around it (and just compile the algorithm selected, so the footprint in the ROM stays small)
- I've looked at tuning options and the following look promising:
  - Ziegler–Nichols
  - Tyreus–Luyben
  - Pessen Integral Rule
  - Ciancone–Marlin
  - Internal-Model Control
  - smooth IMC / Skogestad 
  - Tyreus–Luyben PI
  - Tyreus–Luyben PID
  - Lambda tuning (IMC variant)
  - Ciancone–Marlin correlations
  - and manually setting `kp`, `ti`, `td`
- Further output filters, apart from simple ranges:
  - the `hysteresis` filter allows to wait for consecutive number of "increases" / "decreases" seen before the output is modified.
  - the `hysteresis` filter allows for setting a clamping of the direction of change to a certain time, meaning the output can't oscillate quicker between up and down than specified.
  - the `pacing` filter will allow to do token bucket rate limits. TL;DR you can specify the maximum amount of total movement within time frames. Say you want a relay to just move once in 10 seconds? Set it to `10s: 1`. You want a valve to be actuated just 10% in one minute? just set it to `1m: 10`. This allows also multiple timeframes to be specified.
  - the `duty_cycle` filter is not for actuators but for continues operating devices, like pumps or motors which run for a long time, but have a duty limit for cooling purposes. For them you can set one or multiple duty limits, like `60m: 20m` while also specifying the threshold at which it is no longer counted towards the duty limit, say if you run with very low output.
- in addition there will be a special filter I just made up, which is called "defer_blocked". The idea is, that if anything of the other filters blocks the output, the first value is stored and not output, and is used as temporary clamping limit in this direction. The PID algorithm is run further, but all further outputs are just checked against the temporary clamping limit. So if the output is increased further, this will be ignored, however if the PID backs off and want to increase less, or even decrease, this is stored in addition. If the block then is released, either the clamping limit is pushed out as new output, or the backed off value, depending what is the latest value by the PID. On the next run the imaginary "overcompensation" PID tends to do, if it can't output is then completely reversed, as only the clamped output is fed in as the worst increase which can happen.

Feel free to point out any limitations this will introduce to your use case. Or any suggestions you might have.

I have a look at them before I start implementing.

vontrapp, 2025-07-26:
All great ideas. I love it.

I don't know if this fits into a filter or something else, or if it is feasible at the component level, but regarding quantization:

I implemented a PID on a peltier wine cooler. In order to get a high enough PWM frequency I had to sacrifice resolution on the output and the quantization was severe enough that it induced oscillations in the process that couldn't be tuned out. One level would cause the temperature to climb slowly and the very adjacent output would cause the temperature to fall slowly.

So I combined PWM and sigmadelta together. Let's say it was 100 levels of PWM output for simplicity, 1% increments, but the process output called for 43.6%. The combined output would toggle between 43% and 44% output, with that toggling happening on the period and control of a sigmadelta float output of 0.6.

It worked amazingly well. But it was really complicated to combine those outputs together. A way to abstract that complexity and ask for an increased output resolution by controlled movements between quantized output levels could be really powerful.

Thinking about a name I don't have one yet, but I think the core of the component class is controlling a process with a variable output of more options than just on/off - either discrete or continuous. I like the idea of naming it adjacent to proportional, as the output is set in proportions rather than on/off. Or adjacent to variable output, like a VFD (variable frequency drive). Or something like finesse. Something to do with resolution as the the output has more resolution compared to 1 bit binary. Maybe analog but that implies not discrete output levels. Just some brainstorming material there.

RubenKelevra, 2025-07-26:
@vontrapp what was the period of this oscillation?

patrickcollins12, 2025-07-26:
My version here now compiles... I haven't tested it. Might work as a good starting point?

RubenKelevra, 2025-07-26:
Thanks for sharing this @patrickcollins12, I'll have a look!

fu-zhou, 2025-08-15:
@adamja : in short: yes, fully correct! 

In my specific case I use iobroker with the esphome API wrapped in a driver/ adapter for iobroker, the PID controller is another adapter. And then there is the javascript/ blockly adapter. I use a blockly script to hand over the temperatures (average value) coming from the esp32 rtd-boards as actual value to the PID controller (in iobroker), the set point is sliding based on the meat temperature (just a mathematical calculation in blockly in iobroker) and the controller output is passed on to the esp32 to control the stepper motor via the ULN2003 driver board. Once the wireless connection to the esp32 is/would be lost, I use the binary input of the PID Controller to hold/ suspend the controller, i.e. keeping the output as is to bumplessly resume once the

As there is also an S7 Adapter in iobroker, I could use the blockly script to pass on the values to the S7 plc and use the PID controller there but that would be too much effort. That was also the argumentation why I could motivate the iobroker developers to make a PID controller adapter available - based on the OSCAT PLC library

So having this PLC controller in esphome would be awesome!

PS: here's the link to the iobroker PID:
https://github.com/mcm4iob/ioBroker.pid

fu-zhou, 2025-08-16:
Illumination/brightness inside a room would be a use case. Set point inside the room is e.g. 800 lux. If the sun is shining, the dimmer output is 0%, at night the dimmer is at 100%, if it's overcast, the brightness controller keeps the light at 800 lux by controlling the dimmer, depending on the intensity of the sun.

Air stream of a ventilation unit (m³/h or cubic feet per minute depending on the continent you're in;-)) depending on the season, when bypassing the heat exchanger and therefore the resistance in the duct system changes.

Charging current of your wallbox (electric vehicle) in order to control the solar grid input to 0 kW => grid input is controlled to 0 Watts,

Charging power of your electric heater limiting the grid consumption when dish washer and washing machine are also running. Their power consumption fluctuates quite a bit.

So units I could see being controlled inside a home are:

air:
- temperature & temperature difference
- humidity (relativ&absolute) as well as enthalpy
- flow

water:
- temperature
- flow
- level

el. power:
- current
- power
- brightness

required sensors:
- temperature, humidity
- flow/ pressure (orifice)
- power meter
- brightness meter

possible actuators:
- valve (0-100% or PWM)
- stepper motor
- dimmer
- inverter/ frequency drive also via API (e.g. MQTT) incl. wallbox, solar inverter

The list does not claim to be complete, maybe a start...

clowrey, 2025-08-16:
> Can I ask, does anybody have a use case for a non-temperature pid controller beyond humidity control?

Yes EV charging current control to limit it to available PV power. That is why I am here.

Inkemann, 2025-08-25:
> Can I ask, does anybody have a use case for a non-temperature pid controller beyond humidity control?

I currently use the climate pid entity to match fans to a specific RPM setpoint.

adamja, 2025-09-04:
Hi guys, I'm going to try and create the below into a component. Does anyone have any feedback for what I am trying to achieve with all of the component options? It's grown a lot to add all the features I want to include for the `pid_controller` component, but I think everything there is needed to meet my requirements. I want all the `pid_controller` options configurable via mqtt and also map all the configurations options and I/O to a display, so I need the api calls via `.cpp` for the lambda's also... not sure if I've over complicated it, but i don't know how to add all of the functionality listed any other way 😄 

I think the clamp logic is not required with the forced input scaling... might remove that.

@patrickcollins12 I had a look at your controller and it had dual outputs, one for raising and one for lowering the input value. Does it work with a single output to raise and lower also? I was also curious how the autotune worked. Does it change the values of PID when you run it, or does it just advice what to set them to?

This is what I am planning to implement for the esphome.yaml configuration after importing via external components:

```yaml
external_components:
  - source:
      type: local
      path: components
    components: [pid_controller]
```

### Component configuration:

```yaml
pid_controller:
  id: my_pid                              # required — unique component id

  # IO
  input_sensor: pid_input                  # required — source sensor id (RAW units)
  setpoint: 120.0                          # required — RAW units; normalized using input_scale_*
  output_sensor: pid_output_norm           # required — publishes NORMALIZED output (0..1)
  output_scaled_sensor: pid_output_scaled # optional — publishes RAW scaled output; default: not published

  # Control gains (fixed)
  kp: 1.0                                  # optional — default: 1.0  (normalized domain)
  ki: 0.2                                  # optional — default: 0.0  (normalized domain, sec^-1 scaled by sample time)
  kd: 0.1                                  # optional — default: 0.0  (normalized domain, sec)

  # Normalization (REQUIRED) — maps RAW input → normalized [0..1]
  # n = clamp( (raw - input_scale_min) / (input_scale_max - input_scale_min), 0, 1 )
  input_scale_min: 20                      # required — RAW value that maps to 0.0
  input_scale_max: 200                     # required — RAW value that maps to 1.0 (must be > min)
  output_scale_min: 100                    # optional — default: unset (no scaling). RAW value when output = 0.0
  output_scale_max: 200                    # optional — default: unset (no scaling). RAW value when output = 1.0 (must be > min)

  # Display scaling (optional) — for UI/telemetry only (real-world units; does NOT affect control)
  # These define the axis/units shown in UIs or logs:
  #   input display value   = input_scale_display_min  + n_in  * (input_scale_display_max  - input_scale_display_min)
  #   output display value  = output_scale_display_min + n_out * (output_scale_display_max - output_scale_display_min)
  input_scale_display_min: 0.0             # optional — display value when input is at input_scale_min (e.g., 0.0 °C)
  input_scale_display_max: 180.0           # optional — display value when input is at input_scale_max (e.g., 180.0 °C)
  output_scale_display_min: 0.0            # optional — display value when output is 0.0 (e.g., 0.0 % valve)
  output_scale_display_max: 100.0          # optional — display value when output is 1.0 (e.g., 100.0 % valve)

  # Mode & behavior
  controller_mode: off                     # optional — default: off  (off | automatic | manual) — safe-by-default
  off_output: 0.0                          # optional — default: 0.0  (NORMALIZED 0..1 used when mode=off)
  # manual_output: 0.42                    # required iff controller_mode=manual (NORMALIZED 0..1)

  controller_direction: direct             # optional — default: direct    (direct | reverse)
  proportional_on: measurement             # optional — default: measurement (measurement | error)

  # Sampling
  sampling_mode: event-driven                # optional — default: time-based (time-based | event-driven)
  sample_time: 200ms                       # optional — default: 100ms (min 10ms; 100–1000ms typical)

  # Authority (anti-windup limits inside PID)
  output_clamp_min: 0.0                  # optional — default: 0.0 (NORMALIZED)
  output_clamp_max: 1.0                  # optional — default: 1.0 (NORMALIZED; must be > min)

  # Rail alarm thresholds (telemetry; separate from clamps)
  output_limit_lower: 0.10                 # optional — default: unset/disabled (NORMALIZED; trips “low rail” alarm)
  output_limit_upper: 0.90                 # optional — default: unset/disabled (NORMALIZED; trips “high rail” alarm)

  # Alarm outputs
[... 49 lines cut ...]
  input_scale_min_number: pid_in_scale_min     # optional — default: not bound (uses static input_scale_min). RAW units
  input_scale_max_number: pid_in_scale_max     # optional — default: not bound (uses static input_scale_max). RAW units; must be > min
  output_scale_min_number: pid_out_scale_min   # optional — default: not bound (uses static output_scale_min). RAW units
  output_scale_max_number: pid_out_scale_max   # optional — default: not bound (uses static output_scale_max). RAW units; must be > min

  # Display scaling numbers (optional — UI/telemetry only; does NOT affect control math)
[... cut ...]
  proportional_on_select: pid_prop_on           # optional — default: not bound (uses static proportional_on). Options: measurement/error
  sampling_mode_select: pid_sampling_mode       # optional — default: not bound (uses static sampling_mode). Options: time-based/event-driven

  # Optional: make anti-windup clamps tunable at runtime
  output_clamp_min_number: pid_out_clamp_min    # optional — default: not bound (uses static output_clamp_min or 0.0). NORMALIZED 0..1
  output_clamp_max_number: pid_out_clamp_max    # optional — default: not bound (uses static output_clamp_max or 1.0). NORMALIZED 0..1; > min
```

[11 more replies]