esphome/feature-requests #1872: INA226 alert pin Esphome

tiimsvk, 2022-09-16:


ssieb, 2022-09-16:
This is a valid feature request.

gargamelonly, 2023-11-03:
This is a valid feature request.

JoshHiggins92, 2024-04-26:
This is a valid feature request. I expect to use the alert pin to switch on a relay that will charge batteries, as well as wake up my esp32 from deep sleep. The latter being more important so the device stays in deep sleep and doesn't consume power until a solar cell charges the battery beyond a certain voltage and wakes the esp32 up.

FelixB22, 2024-05-12:
I miss this function also.
I want to set the lower and higher alarm position for esphome
thanks

gargamelonly, 2025-12-27:
Someone implemented the ALERT pin in a custom library: 
[https://github.com/RobTillaart/INA226](https://github.com/RobTillaart/INA226)
It can be a good reference for adding it into ESPHOME.