esphome/feature-requests #1873: Add water meter readout to DSMR digital meters component

henklietaert, 2022-09-16:
**Describe the problem you have/What new integration you would like**
Add water meter readout to DSMR digital meters component

**Please describe your use case for this integration and alternatives you've tried:**
The water companies in Belgium are doing a trial with digital water meters that connect to the electricity meter to send out the messages. I assume the values can be read through the P1 port.
This would be handy to follow-up water usage as well as detect any leaks due to excessive consumption.

**Additional context**
I have no idea how to start this integration, but I'm wiling to test and provide information if needed to help developing this. I have a digital water meter installed since Monday.

bekovan, 2023-02-24:
Hello,

I captured a telegram via the debug option

Following lines seems to describe the watermeter :
"
0-2:96.1.1()
0-2:24.2.1(230224135019W)(00045.409*m3)
"

2023-02-24 13:50:15.256 DEBUG (MainThread) [dsmr_parser.clients.protocol] got telegram: /FLU5\

1-0:1.8.1(000598.038*kWh)
1-0:1.8.2(001358.258*kWh)
1-0:2.8.1(000984.518*kWh)
1-0:2.8.2(000331.240*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.000*kW)
1-0:1.6.0(230222200000W)(02.504*kW)
0-0:98.1.0(2)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221225130000W)(03.850*kW)(230201000000W)(230108194500W)(03.636*kW)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.597*kW)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.010*kW)
1-0:22.7.0(01.608*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:32.7.0(235.3*V)
1-0:52.7.0(000.0*V)
1-0:72.7.0(236.2*V)
1-0:31.7.0(007.08*A)
1-0:51.7.0(007.94*A)
1-0:71.7.0(001.14*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1()
0-1:24.4.0(1)
0-1:24.2.3(230224135001W)(01147.922*m3)
0-2:24.1.0(007)
0-2:96.1.1()
0-2:24.2.1(230224135019W)(00045.409*m3)
!A840

Nielsvanvlierberghe, 2023-02-26:
Yes, correct.
This is what's mentioned from following document:
https://maakjemeterslim.be/rails/active_storage/blobs/eyJfcmFpbHMiOnsibWVzc2FnZSI6IkJBaHBBZ0lEIiwiZXhwIjpudWxsLCJwdXIiOiJibG9iX2lkIn19--cdd9b48fd0838e89b177f03b745b23450fd8f53e/e-MUCS_P1_Ed_1_7_1.pdf?disposition=attachment

[image]

Would love to see the integration of the sensor.

bekovan, 2023-04-05:
A workaround I implemented today, basically with some equipment I will need for the installation of an EV and a slimmelezer I had still collecting dust 

I noticed that the watermeter installed in my house is sending the measurement in the same register as used in the Netherlands for the gas measurement, so -->

1) connect P1 port of the power meter to a P1 splitter ( used : active P1 splitter from HomeWizard)
2) connect  slimmelezer 1 with Esphome --> use this for electricity & gas in the energy dashboard
3) flash slimmelezer 2 with Esp-link
4) connect slimmelezer 2 in Home Assistant with DSMR Slimmelezer, use DSMR 5 (not 5B)
5) create a new variable renaming the gas output from slimmelezer 2 to device class water (if you would like to use this in the energy dashboard)

Not elegant but functioning

a1ad, 2023-10-28:
Here is someone that worked on it: https://github.com/peno64/slimmemeter/wiki

Can we create an official implementation? It's the one thing missing!