esphome/feature-requests #1878: Allow publishing dignostic sensors as attributes

SergioRius, 2022-09-18:
**Describe the problem you have/What new integration you would like**

It would be better if we could publish all the diagnostic info as HA attributes, for example under status sensor. I have a base yaml file that creates the status sensor and then ip, rssi, uptime, vcc, version.
All those sensors are created and thus generate recordings for every esphome device I have.
Also, when I search for my light, or for the living room lights in the ledge (3), for example, loads of entries appear in the result. It would be better, for cleanliness at least.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-09-18:
Duplicate of https://github.com/esphome/feature-requests/issues/1635

nagyrobi, 2022-09-18:
Also see: https://github.com/esphome/feature-requests/issues/1112

closed 2022-09-18 by nagyrobi