esphome/feature-requests #1879: Sprinkler controller - add new public method instead of protected

timota, 2022-09-18:
**Describe the problem you have/What new integration you would like**

Make protected method `reset_resume_()` in Sprinkler Controller `Public`.   

**Please describe your use case for this integration and alternatives you've tried:**

Consider following scenario: 
  Requirements: 2 buttons (1: start individual zone1, 2: pause/resume). 
1. User pressed start `Zone1`
2. User pressed `Pause`
3. User pressed `Zone1` - in this case controller will start new cycle for Zone1 without restoring paused state. 
4. Zone1 cycle in a middle - User pressed `Resume` - controller will shutdown current cycle and resume stored aka paused state. 

It can be classified as a bug, but from another point - it can be a feature, where new public method aka `reset_resume` can be called in step 3. So user can reset paused state and start normal Zone1 cycle.

**Additional context**

nagyrobi, 2022-09-19:
@kbx81

kbx81, 2022-09-19:
Seems reasonable -- that method probably should've been made `public`, anyway. 😄

closed 2022-09-20 by jesserockz