esphome/feature-requests #1880: BLE Tracker Proxy MAC based sensor

atheon25, 2022-09-21:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-09-21:
This already works like you'd like to. But please clarify:
- What's an _Atom BLE-Tracker-proxy_?
- How do you configure your RuuviTags? In ESPHome, or directly in Home Assistant?

nagyrobi, 2022-09-21:
Did you see there's a HA custom component for RuuviTags?
https://github.com/ruuvi-friends/ruuvi-hass.io

atheon25, 2022-09-21:
Yes I saw that, but i'm notsure how it works with BLE-tracker-proxy.
Those BLE-tracker proxy are crucial because many of those RuuviTag is far from Home assistant device.

nagyrobi, 2022-09-21:
Can't help unless you provide answers

atheon25, 2022-09-21:
[Atom BLE-tracker-proxy](https://esphome.github.io/bluetooth-proxies/)

nagyrobi, 2022-09-21:
Not clear what are you referring to as _Atom_.

But anyway, you shouldn't configure the tags in ESPHome at all. To use the benefits of the new proxy functionality, you have to configure them in Home Assistant directly.

atheon25, 2022-09-21:
Atom I mean M5Stack Atom.
So you mean in ESPHome I should have only ruuvi_ble configured and ruuvi-frend confiquration my configure.yaml

Let me try that....

nagyrobi, 2022-09-21:
No. Delete entirely the ruuvi_ble from ESPHome node proxy.

atheon25, 2022-09-21:
OK it work but only partially.
ie. RuuviTag humidity in gauge card give me Entity is non-numeric: sensor.storage_humidity

am I still do something wrong?
I reprogram ESP32 with proxy software from ESP home bluetooth-proxy and didn't add anything.
I also add ruuvi-friends type of configurations to configuration.yaml
Now I see all entities in list but not get those working in any standard or custom card proberly.

nagyrobi, 2022-09-21:
You have to raise that as issue to the custom component author.

closed 2022-09-21 by nagyrobi