esphome/feature-requests #1883: "Window open" support for thermostat

oxc, 2022-09-21:
**Describe the problem you have/What new integration you would like**
I would like a way to immediately turn off my `thermostat` heating even if it has not reached `min_heating_run_time`.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to have a `min_heating_run_time` configured, to prevent cycling between on/off too often (especially since the heating valve motor has its own reaction delay time).
However, at the same time, I would like to configure an automation that immediately turns of the heating if the window in that room is opened. This should not wait until the `min_heating_run_time` has passed.

After the window has been closed, ideally I would like to immediately start heating again if necessary, even if the `min_idle_time` or `min_heating_off_time` has not passed.

**Additional context**
I could imagine this as a specific feature of the thermostat component, or as a separate preset. Allowing presets to specify different `min_x_time` values would most likely suffice for this use case. Of course, the preset switching logic would become a bit more complicated then.

nagyrobi, 2022-09-22:
This could be achieved by placing conditions in the actions of the thermostat, based on the window open sensors.

ssieb, 2022-09-22:
That wouldn't do anything.  The actions won't get called until the run times expire.
I don't think this is a useful request, but it is a request, so unless someone else vetoes it, it can stay open.  If you need the min run time, then why would you want a way to bypass it?  And is it really that long that it's likely to matter?
The amount of changes and code involved to implement this is not worth the extremely limited use case.

nagyrobi, 2022-09-22:
Combined with triggers from the window open binary sensors, which could act on the heating source directly, +conditional actions in the thermostat, this can be done without an source code mods, all in config.

oxc, 2022-09-22:
Well, my min heating time/off time is set to 5 minutes which I copied from the examples. I might be able to reduce that in order to get closer to my desired state, at the same time increasing the possibility of wear on the thermal switches.

Or, act on the hardware switches directly, yes, circumventing the thermostat completely.
Both seems a bit icky, to be honest.

The fact that something "can be done without any source code mods, all in config" does not mean changes are not warranted at all.

oxc, 2022-09-22:
But, of course, thank you for your input. I will try your suggestions!

If it raises further problems, I will try to find a change to the thermostat code that is minimal but allows me to reach my goal. For example, having a version of setPreset that allows cancelling existing timers could work.