esphome/feature-requests #1884: Integrate `pvvx mithermometer display` into bluetooth_proxy

JeeCee1, 2022-09-22:
**Describe the problem you have/What new integration you would like**

Since the great `bluetooth_proxy` functionality I have deleted my 30ish `LYWSD03MMC` devices from the esphome configs and have them discovered directly in HA.

This display feature opens up a lot of great possibilities but as I understand it it requires to make a local config on the ESP device, defeating the `bluetooth_proxy` functionality.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-09-22:
No, this is not possible. The scope of  `bluetooth_proxy` is to behave much like a Bluetooth adapter plugged in HA, only that it's not USB, but extended via network instead. Data passing through the proxy is not interpreted by the node. Moreover, data can be encrypted via a `bindkey` (which, for security considerations, is recommended to be used), and the decryption is done in HA, not in the proxy. The proxy doesn't even know the key, it just forwards the packets (regardless if data is encrypted or not).

So if you want to achieve the same _display_ functionality but through `bluetooth_proxy`, you have to re-implement the whole thing at HA level, for example via a custom component for HA.

But keep in mind, that at the moment `bluetooth_proxy` is still unidirectional, it just forwards passively the broadcasts of BLE devices to HA. HA can't establish a bidirectional communication yet through the proxy, eg. can't pair with the device through the proxy.
Additionally, there's still the issue that if you have multiple proxies and a device is in the range of both, currently both forward the BLE broadcasts into HA, and HA uses them both. But when bidirectional communication will be implemented, that will not be easily possible, because a device will still be able to pair with one node only, transferring the pairment from one proxy node to the other, if a device is in multiple ranges is not something that can be easily done => with pairing it's likely that you'll still be tied to one of the proxies, not multiple ones => meaning that in that case you're still better with the current display implementation as it is, via ESPHome.

nagyrobi, 2022-09-22:
However, according to pvvx display docs:

> After the data has been transmitted, the BLE connection is terminated in order to be able to receive the advertising data required for the pvvx_mithermometer sensor platform

This should mean that adding the display component to the ESPHome node would not necessarily break your BTHome integration in HA. Try keeping the display component on your ESP32, only delete the sensors, and leave the proxy and the tracker set up with scan parameter `active: false`. That way, proxy shouldn't interfere with the display part...