esphome/feature-requests #1886: Call ESPHome native API via Arduino

tbaumann, 2022-09-22:
**Describe the problem you have/What new integration you would like**

It's quite easy to integrate Arduino libraries and run code via lambdas. But it would be nice to also have it the other way around.

I have an extensive Arduino project to which I would like to add Home Assistant features. (publish sensor data, perhaps expose a switch or service)

I know MQTT exists. But it would be really nice if I could just call the native API from my project.

**Please describe your use case for this integration and alternatives you've tried:**

MQTT. But the words from https://esphome.io/components/api.html?highlight=api#advantages-over-mqtt are very true.

**Additional context**

RoboMagus, 2022-10-28:
Do you have any concrete examples to share? e.g. the way you are integrating ESPhome within your extensive Arduino project?

From my point of view it's quite unlikely that ESPHome sources are compiled outside of the regular ESPhome environment. Assuming you instantiate your Arduino code using a custom component / lambda, or the like, you should be able to pass the sensor objects you'd like to interact with using those code snippets.

tbaumann, 2022-10-28:
I don't mean to integrate  ESPhome within another project. Just the native API so I can expose elements into HA.

My example is a coffee machine. An ESP with arduino code controls the temperature and brew functions. I could expose the device status and temperature as HA sensors.